breadbee/breadbee #2: PoE capability

StKob, 2020-04-14:
Are there any plans to build support for PoE? This board could be used for quite many applications where it could be powered via ethernet plug while communicating. For example: diy weather sensors.

fifteenhex, 2020-04-15:
I think getting the transformer and HV step down onto this board would be pretty hard. There isn't a lot of space and it was already a pain to make it this small and within the bounds of what the cheap Chinese PCB fabs can produce.

StKob, 2020-04-15:
Could you actually make board 32 x 32 and fit that stuff? Although I don't know if that would make big price difference.

fifteenhex, 2020-04-15:
I don't think the additional 2mm helps that much. At 4 layers without expensive stuff like blind vias (vias that are only present on some layers not through the entire board) there isn't much more space to route anything.

Looking at commercial PoE modules (i.e. https://www.alibaba.com/product-detail/12w-Output-Power-Over-Ethernet-PoE_112394905.html) I think it'd be physically impossible to get it onto the board without making it twice as big.

If there was an RJ45 jack that had the ethernet magnetics and PoE parts cramed into it then maybe it'd be possible to swap the RJ45 jack over. I don't see any other way it could work.

StKob, 2020-04-15:
Something like this? 
https://www.te.com/global-en/products/connectors/intersection/industrial-rj45/industrial-rj45-integrated-magnetic-jacks/ethernet-jacks-with-integrated-magnetics-and-poe.html

igut, 2020-04-15:
Using a POE capable Ethernet PHY you could extract the power from the cable to 4 additional pins from where an external POE module could provide power to many different applications and to the device itself. I have in mind this kind of modules: http://www.r-tone.com/en/ProductList.asp?SortID=35 The same approach can be seen on Banana-pi,  Rock-pi etc. This would make the BreadBee desirable for a broad range of applications!

fifteenhex, 2020-04-16:
I understand what you guys as saying. This is my opinion on it:

- If you can find a RJ45 jack that has the magnetics and the PoE bits fully integrated that is the same size as the current one I can see if it's possible to make that work somehow. The problem is that most RJ45 jacks have pins that go through the board and there is no room in the current design for anything else to go through the board as both sides are fully packed. The current jack was the only SMD jack I could find that was cheap and didn't need cut outs in the board.

- If you want a board like this with PoE I would get some of those solder in modules, take the breadbee schematic and make a little board that solders right onto the back of it. If you don't mind the older version of the chip used here that runs at 800mhz and not 1ghz you can get them for about $2.5 or I can give you a few.

StKob, 2020-04-16:
I've found something like this: http://www.link-pp.com/upload/file/20190307/20190307135160306030.pdf

fifteenhex, 2020-04-16:
That doesn't include the electronics to actually do the PoE part.

StKob, 2020-04-16:
Ok, but you can then provide pins from that. Then it would be possible to make optional daughter board.

fifteenhex, 2020-04-20:
> optional daughter board.

I think once you have a daughter board you might as well use an in-line PoE splitter with a microusb output.

sbridger, 2020-04-23:
> I think once you have a daughter board you might as well use an in-line PoE splitter with a microusb output.

I disagree. Splitters are neither small, or cheap (relatively). Having the transformer centre taps available to feed an external circuit is ideal for me.
1) No added cost when it's not needed
2) You are not committed to any particular way of doing poe. The standards compliant way, is not what many hackers would be doing. (eg putting 4 breadbees in a box, you just feed 5V down the poe)
3) possibly putting a bridge rectifier / footprint on the breadbee would allow any little smps to be wired up. The bridge is not on many little smps boards.

fifteenhex, 2020-04-23:
>Having the transformer centre taps available to feed an external circuit is ideal for me.

The center taps of the transformer are available on C33 and C34.

> 3. possibly putting a bridge rectifier / footprint on the breadbee would allow any little smps to be wired up. The bridge is not on many little smps boards.

I don't think there is enough space to do that and keep the trace width and spacing large enough that it can be produced on the cheap pcb services without making a mess.
I think there might need to be a bit of consideration on how to stop the PoE supply from back feeding into the usb too.

If someone whats to doodle out what is needed. How it'll fit etc I'm happy to try to add it.

sbridger, 2020-04-24:
OK, I will have a look.

btw , (on schematics) "M" is used for megohm resistors, "m" is milliohm.

Might be an idea to have RC filter on AVDD_node

On sch, I see you aren't taking the usb out to the usb connector.

Perhaps an option would be to put CH330, (8 pin SOP) over the usb resistors to USB_d?

Thus the usb could connect if resistors are in, or a CH330 could be fitted (easy to solder/remove sop8)
These [adaptors](https://www.aliexpress.com/item/32814218082.html) could be used for usb master

fifteenhex, 2020-04-24:
>btw , (on schematics) "M" is used for megohm resistors, "m" is milliohm.

mmm ok. I'll update those values.

>On sch, I see you aren't taking the usb out to the usb connector

Initially I didn't expect to be able to reverse engineer the usb part so I didn't bother.
I'm not sure if a CH330 would fit. I think a SOP8 is bigger than a MSOP10. But I think there could be some jumpers somewhere to connect the usb data lines to the on board usb port.

fifteenhex, 2020-05-19:
Closing this for now.

closed 2020-05-19 by fifteenhex

moqmar, 2021-04-21:
Also interesting would be support for 5V passive PoE, that should be a lot easier to implement?

fifteenhex, 2021-04-22:
> Also interesting would be support for 5V passive PoE, that should be a lot easier to implement?

If you want to do the schematic changes for it I don't mind trying to route it in the PCB.

moqmar, 2021-04-23:
Oof, seems like the [HR961160C](https://datasheet.lcsc.com/lcsc/1811141110_HANRUN-Zhongshan-HanRun-Elec-HR961160C_C55683.pdf) more or less just grounds the spare pins and thus doesn't expose the neccessary pins for any kind of PoE...

So, I guess without changing the RJ45 connector, PoE is not gonna happen...

moqmar, 2021-04-23:
[Abracon ARJ-102S](https://abracon.com/Magnetics/lan/ARJ-102S.pdf) and or [Molex 0938098046](https://static6.arrow.com/aropdfconversion/4a4189b549f64118f5c998b09980b07fb6c595a7/0938098046_modular_jacks_plug.pdf) should work, but they cost at least twice as much.

fifteenhex, 2021-04-23:
> Edit 3: it seems like both are protruding through the PCB, so I guess that's not really feasible for this project? Or can we raise the PCB somehow by soldering another one onto it for that part...

That's the main issue I think. The current connector is the only one I could find that is SMD (can't have pins through the board) and doesn't need a cutout in the board. I think an SMD connector without the transformers + an external SMD transformer packager would work but I think it would be too long.

moqmar, 2021-04-23:
Soo, the old connector is 26.6mm long (including pins). Most transformer packages (like the TG110-RPE5N5LF) are 9.53mm long (including pins), and there's the TE 6339191-1 or some others being 16.48mm long (including pins), yielding 26.01mm (which isn't achievable as the pitch doesn't match - I couldn't find a 1.27mm pitch jack with LEDs yet).

If we can then fit some tiny connector for pins 4/5, 6/7, VCC and GND, those could be bridged for passive 5V PoE, or converted with a tiny daughter-board with a RPMH5.0-1.5 for higher-voltage passive PoE, or even 802.3af Mode B PoE (requiring some additional resistors and).

fifteenhex, 2021-04-24:
>  I couldn't find a 1.27mm pitch jack with LEDs yet).

Without LEDs works.. the leds on the current connector aren't connected as the chip doesn't expose pins to drive them.

moqmar, 2021-04-24:
If we can give up on the shielding and CAT 3 is enough as well then there's the 14.5mm long Amphenol 73304-111LF. I'm not sure though if that's such a good idea...

Then there's the Amphenol RJSSE-5380 which is 18.29mm long including pins, but the pins are 2.57mm long so they can be shortened a bit - it's more of a hack though then.

moqmar, 2021-04-25:
[Würth 7498210002](https://www.mouser.de/datasheet/2/445/7498210002-1723983.pdf) 😀

It solves all of our problems, but holy shit is it expensive, almost 9 €...