breadbee/breadbee #3: Battery, Solar, & Native Low Power Mode support

PracticalCode, 2020-04-14:
Are there any plans to build support for a battery 'hat'? 

And\or are there any plans to build support for a solar 'hat', which could perhaps recharge the battery? 

Along those lines, plans to support the low power mode (allowing the unit to 'sleep', and wake once an hour, then back to sleep.)

This board could be used for quite many applications where it could be used for remote \ standalone operations. For example: diy weather station, doorbell, security 'nanny' cam (especially once you get the camera supported.)

fifteenhex, 2020-04-15:
>Are there any plans to build support for a battery 'hat'?

There are plans for the Wifi enabled version to have a proper PMIC that has a charger etc.
The issue is getting a PMIC with the right default voltages as all of the very cheap ones (AXP, Injoinic) are apparently ordered in 2000+ batches with the customers required default voltages set and the ones I can get have the wrong voltage for the DDR memory. 

With that said I have also looked at possibly using the same 3 in 1 DC-DC buck chip with a lipo charger with power path (see https://github.com/fifteenhex/dbmbyhd)) as an alternative as I don't want to keep having to build 2000 of things.

>And\or are there any plans to build support for a solar 'hat',
>which could perhaps recharge the battery?

I have no plans to do that myself but there's no reason someone else couldn't. If you don't need mppt etc I think just having a proper PMIC with a good input range would work. If someone wants to detail want is needed to do it right I can try to integrate it into the wifi version.

>Along those lines, plans to support the low power mode (allowing the unit to 'sleep',
>and wake once an hour, then back to sleep.)

The chip apparently contains everything to go into DDR refresh only with the CPU turned off that should achieve what you want. The RTC is listed as a wake up source in the register descriptions I can find too so potentially you could set an RTC alarm. I have managed to get the CPU to turn off and it's power consumption is ~10mA like that. I haven't managed to get it to turn back on yet.
Also this version of the board has no control over the voltage rails. A proper PMIC would allow the core voltage to be scaled and disabled totally when sleeping which should bring it down a bit more.

>doorbell, security 'nanny' cam (especially once you get the camera supported.)

For camera applications IP camera boards with the same family of chip are much cheaper:

https://item.taobao.com/item.htm?spm=a230r.1.14.27.68d32e0et9vGsD&id=562051542010&ns=1&abbucket=17#detail

The uboot/kernel for this board runs on those modules. However I wouldn't hold your breath on camera support. I've worked out where a lot of the hardware is and how it's sort of plugged together but actually doing something with it is another story.

juehv, 2020-04-16:
I'm highly interesed in a SBC with suspend to ram capabilities, too :)
That would be very nice for my battery powered projects!
Looking forward to your crowd funding campain 👍

fifteenhex, 2020-04-26:
Update on this:

Suspend to ram is now working to the point that it suspends and the RTC is able to wake it back up via an alarm but the boot rom gets stuck on reentry. My usb current meter shows 0.00A when it's sleeping so the consumption must be pretty low.

fifteenhex, 2020-04-26:
Another update:

Working to the point that the kernel is restarting now. Hangs just after. Needs some love to work out what is locking it up but pretty close.