breadbee/breadbee #4: Development setup

kadamski, 2020-04-15:
What development setup do you have for the board? My understanding is that the ethernet works in u-boot so the kernel and rootfs can be loaded over tftp/nfs, right?

The IPL, u-boot SPL and u-boot itself must be on the SPI NOR currently and when changing u-boot this could be broken it has to be desoldered, reflashed and soldered again or there is some easier option? Or can u-boot proper be loaded from JTAG so only IPL/SPL need to be on the flash?

fifteenhex, 2020-04-16:
Sorry for the brevity. Writing this while eating on my lunch break..

>What development setup do you have for the board?

tftp for the breadbee itself. The other chips have a bigger bootrom that can load a blob from sd card so I have combined uboot + kernel blob that gets loaded into memory and then I tell uboot to bootm the image thats in memory.

>My understanding is that the ethernet works in u-boot so the 
>kernel and rootfs can be loaded over tftp/nfs, right?

Yes. the SD card doesn't work in uboot yet so ethernet or loading over serial are the only options.
The usb ip already has a driver in uboot so that should also work with some hacks.
The current setup gives you a rescue image with a really small kernel and rootfs that is just enough to reload the main image from network, usb or sd.

>when changing u-boot this could be broken it has to be desoldered, reflashed and soldered again

As long as you don't break the IPL and the u-boot SPL image breaking u-boot isn't a problem as the SPL is configured to be able to load the main uboot over serial. That has saved me a few times.
For times I have broken the IPL/SPL I have used a clip to reflash the chip in place. It's far from ideal.

Long term the bootrom seems to have a UART boot mode. Working out how it works and writing a flashing program are on my todo list.

kadamski, 2020-04-17:
> The current setup gives you a rescue image with a really small kernel and rootfs that is just enough to reload the main image from network, usb or sd.

Could you explain a little bit how this rescue image works? How can we choose to boot from it instead of the main one? This is meant to be used when kernel/rootfs does not start, right? There is no such rescue for broken u-boot?

> the SPL is configured to be able to load the main uboot over serial.

What is the protocol for that? How to enter this boot mode?

How about JTAG? My understand is that u-boot SPL enables it so it should be possible to load u-boot via openocd, isn't it?

kadamski, 2020-04-17:
Also, is https://linux-chenxing.org/isp/ somehow related? My understanding is that there is some i2c -> spi bridge to flash the spi-nor but I did not understand how to use that. Could that be used for recovering broken flash? Is there some dedicated i2c bus for that? Does it always work or it has to be enabled somehow? Do you know any details on that?

fifteenhex, 2020-04-17:
> Could you explain a little bit how this rescue image works? 

u-boot will stop autoboot if you type "bzzbzz". From there you can load a kernel from tftp or run bb_boot_rescue and have it load the rescue environment. It would also be possible to have it load if a gpio is grounded or something.

>There is no such rescue for broken u-boot?

If the u-boot image is broken in flash the SPL will complain it couldn't load it and wait for uboot to be sent via serial using the ymodem protocol. Recovering from a u-boot image that works but crashes isn't possible at the moment but it could be implemented with a gpio to tell the SPL to always wait for ymodem.

>How about JTAG? My understand is that u-boot SPL enables it so it should be possible to load u-?>boot via openocd, isn't it?

Yes, that worked last time I tried it. You can load uboot proper into memory and jump to it.

fifteenhex, 2020-04-17:
> Also, is https://linux-chenxing.org/isp/ somehow related? 

All mstar chips seem to be able to become an i2c slave on pm_uart that allows programs to be loaded into memory, registers poked etc. I have the dongle and the windows tool but I haven't needed to use it yet. It should be possible to use that to load a program into memory to flash a blank device or recover a bricked one.

I think working out how the UART load in the bootrom works is the best solution as it wouldn't need special hardware tools.

kadamski, 2020-04-17:
> I think working out how the UART load in the bootrom works is the best solution as it wouldn't need special hardware tools.

But for the i2c method, there is an existing tool so the i2c communication it uses could be sniffed and we could understand how that works. How can UART loader in the bootrom be worked out? Do you have some knowledge about it?

fifteenhex, 2020-04-17:
> But for the i2c method, there is an existing tool so the i2c communication it uses could be sniffed

That's true and other people have sniffed the i2c protocol used by other mstar parts so it's possible it'll be the same. With the current board it's not easy to test though as the usb->uart is on those pins.

>How can UART loader in the bootrom be worked out? Do you have some knowledge about it?

Disassembling the bootrom. I did this to find out how to boot from SD card on the other chips.
It would be mostly a case of finding where the bootrom reads from the UART and working out the path to work out what it wants to see. I'm building some tools (https://github.com/fifteenhex/noriflex) so I can easily mess around with the NOR flash to allow me to more easily work on the boot up parts. I intend to replace the mstar IPL binary as well at some point so everything is opensource.

kadamski, 2020-04-17:
> That's true and other people have sniffed the i2c protocol used by other mstar parts so it's possible it'll be the same. With the current board it's not easy to test though as the usb->uart is on those pins.

Ach, I see. Maybe it would be good to have at least some test points added for that on the future revisions of the board, then? However it should still be possible to solder two wires to the ch340e even now. So the console_tx and console_rx is used as SDA/SCL in this mode?

I still don't understand how to enable this i2c on those pins, though.

>  I'm building some tools (https://github.com/fifteenhex/noriflex) so I can easily mess around with the NOR flash to allow me to more easily work on the boot up parts.

That's cool. What do you plan to put on the other side? Something that will emulate the SPI-nor?

fifteenhex, 2020-04-17:
> Ach, I see. Maybe it would be good to have at least some test points added for that on the future

Yep. I learned about the tool after the board design was done and ordered.

> I still don't understand how to enable this i2c on those pins, though.

I think it's done by pulling one of the pins down on reset to disable the uart and switch in the i2c slave it has.

>That's cool. What do you plan to put on the other side? Something that will emulate the SPI-nor?

There will be a mux chip that can switch an spi nor chip between the target board and a programmer. So I can mess modify the nor flash, switch it to the target, test it, modify it again,..
It'll make testing a lot easier.

fifteenhex, 2020-04-18:
I wired up the uart to i2c on another chip in the series and the i2c looks exactly the same as what other people have found:

kadamski, 2020-04-19:
I was looking in the documents and forums to understand how this i2c debug mode could be enabled. It couldn't be simpler, though. Apparently the UART peripheral is detecting I2C START condition without switching modes, quite cool. My understanding is that UART_RX is SCL and UART_TX is SDA, right?

The ch340e does not have a pull-up on TX line, I think. Does it allow any bitbanging? I know the ch341 does support i2c but on other pins than UART so that would be tricky.

fifteenhex, 2020-04-19:
>My understanding is that UART_RX is SCL and UART_TX is SDA, right?

I'll check. I plugged in the wires and flipped them until i2cdetect saw something. The flashrom tool already has a driver for the mstar spi over i2c which it used. I managed to read out the contents of a chip with it yesterday.

> The ch340e does not have a pull-up on TX line, I think. Does it allow any bitbanging? 

I don't think the ch340e does bit bang but I think it could be faked by using RTS to pull the line low to create the SCL signal and the CTS pin to read the SDA line. The transmit parts of i2c would need to be done by sending out bytes from the uart and getting the SCL timing right to frame the byte as a low or high bit.

It would be very tricky and slow but it would only ever be for recovery, wouldn't need any new hardware etc.

kadamski, 2020-04-20:
> I'll check. 

That is how the brief describes it:
```
Signal Name | Signal Type | Function                                                  | Pin Location 
UART_RX     | Input       | Debug Port for UART Receiver *or Slave I2C Serial Clock*  | 18 
UART_TX     | Output      | DebugPort for UART Transmitter *or Slave I2C Serial Data* | 19 
```

fifteenhex, 2020-04-20:
Well spotted. I haven't looked at it in ages.

I don't think I'll get a response but I've asked the ch340e maker it has a bit bang mode or anything.
I'm not finding an alternative FTDI or otherwise chip that has documented bit bang and is the same size.

kadamski, 2020-04-20:
Btw, it seems like the Conn_01x21 could be extended by 2 pins and could have access to raw console_tx and console_rx pins so that some cheap external i2c controller (or even other breadbee) could be easily used for recovery. Unless you have some specific reason to keep this connector exactly 21 pin wide.