breadbee/breadbee #5: FLASH sizes

lal12, 2020-04-15:
I wondered a bit about the FLASH size. A 16MB flash seems to be a bit small to me. After uboot and kernel it's only 12.5MB, which I would guess is just enough for a base system. Even a minimal embedded linux base system (busybox + ulibc ...) can quickly take up a significant amount of that. So it depends a bit on what game you want to play but for a more RPI and ease of use for anything direction it should have at least 64MB, probably would be better. I would imagine many people would like to work with PHP or Python on it. NodeJS probably won't work anyway with less than >=80MB RAM. which would very quickly go beyond the 12MB rootfs size.

Since this is just a first dev board it might not be that important, but if there is not enough FLASH to develop for such more Higher Level languages/tools it might be a bit annoying later. Though do I understand correctly that the SoC supports an SD Card interface?

fifteenhex, 2020-04-16:
Sorry for the brevity. Writing this on my lunch break...

Currently 16MB NOR gets you uboot, one small kernel + uclibc based rescue environment, one bigger kernel + glibc based environment with python3. There are two issues with going above 16MB; 1 - If the addressing mode gets switched and the kernel crashes it'll probably get stuck at reboot. The AllWinner V3s has this problem. I suspect this chip also has this problem but I haven't tested it. 2 - The memory mapped part of the memory is only 16MB. In the kernel it's possible to revert to CPU driven access for the non-memory mapped part but the performance will be worse for anything in the bit after 16MB. For example loading python3 from flash with CPU access is ~4s and the CPU is pegged. With DMA memory mapped access it's around 2s and the CPU clock isn't ramped up.

SD card works and there is a little wing board for the fine pitch header that has an SD socket, USB socket and gpio breakout. I think for people that need more storage maybe one of those NAND with an SD card interface chip like the Rock Pi S is using is a good solution.

kadamski, 2020-04-17:
@fifteenhex For the first problem, some of the chips support 4byte opcodes which does not require mode switching so there is no problem with reset. I mean the SPI_NOR_4B_OPCODES flag in the kernel. This, however, would require CPU driven access for the flash, I guess, as the memory mapped peripheral does not probably support those opcodes?

fifteenhex, 2020-04-17:
I'll take a look at the registers for the flash controller again. It's possible to set the access opcode so maybe we can trick it into mapping different 16MB blocks into the memory window so it's all memory mapped. I didn't look too hard into it because I thought it would be a minefield of issues with the boot rom.

lal12, 2020-04-17:
Two off topic questions:
- Is there an (inofficial) datasheet for the SoC? I couldn't find anything yet.
- Regarding the "Get in touch!" part in the README how to contact you? EMail on your github profile?

fifteenhex, 2020-04-17:
> * Is there an (inofficial) datasheet for the SoC? I couldn't find anything yet.

linux-chenxing.org is where I'm collecting info. There is one complete datasheet for a different chip that has many of the same registers etc. Aside from that it's a case of trawling through the vendor code

> * EMail on your github profile?

Yep.

closed 2020-05-19 by fifteenhex