breadbee/breadbee #6: Boards for sale?

ConnorRigby, 2020-04-17:
I'm interested in the breadbee project. I help run an open source, embedded linux framework called nerves (https://www.nerves-project.org/ , https://github.com/nerves-project) i'm very interested in the breadbee and would like to know if it's possible if i can buy a board off of you in an effort to get a port of our stack up and running on it. I see that you have a want for a nice python library that i could also help get started.
Thanks for your time.

fifteenhex, 2020-04-18:
Your project looks interesting.

Boards aren't for sale yet. My freebies for people that will help out offer still stands but I'm out of PCBs at the moment. I've ordered more but with the current situation it might be a little while.

AdamKeher, 2020-04-25:
Do you know where I can purchase a few MSC313E SoC IC's?

I want to spin my own version of the board with some additional peripherals but can't see any options other than trying to lift them of existing boards.

fifteenhex, 2020-04-25:
There are still a few sets of 5 for $10 on ebay.

AdamKeher, 2020-04-25:
Could I hassle you for a link. I've searched every MSTAR item on ebay and it seems to be filtering them out in Australia. I can find other MSTAR IC's but no MSC313E.

fifteenhex, 2020-04-25:
Sure:

https://www.ebay.com/itm/5pc-SMT-QFN-IC-chip-MSC313E/153876332882?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D20160908105057%26meid%3Dca4d2bac12314e7ba3089c014104165b%26pid%3D100675%26rk%3D2%26rkt%3D15%26mehot%3Dlo%26sd%3D174233257925%26itm%3D153876332882%26pmt%3D0%26noa%3D1%26pg%3D2380057%26brand%3DUnbranded&_trksid=p2380057.c100675.m4236&_trkparms=pageci%3A430d268a-86fb-11ea-b9eb-1e04fc6b3716%7Cparentrq%3Ab1978a311710a4b7fd181f4affccc675%7Ciid%3A1

https://www.ebay.com/itm/5pc-SMT-QFN-IC-chip-MSC313E/153876332882?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D20160908105057%26meid%3Dca4d2bac12314e7ba3089c014104165b%26pid%3D100675%26rk%3D2%26rkt%3D15%26mehot%3Dlo%26sd%3D174233257925%26itm%3D153876332882%26pmt%3D0%26noa%3D1%26pg%3D2380057%26brand%3DUnbranded&_trksid=p2380057.c100675.m4236&_trkparms=pageci%3A430d268a-86fb-11ea-b9eb-1e04fc6b3716%7Cparentrq%3Ab1978a311710a4b7fd181f4affccc675%7Ciid%3A1

AdamKeher, 2020-04-25:
Thanks, that worked.

closed 2020-06-06 by ConnorRigby