breadbee/breadbee #12: RTS/CTS for serial ports

sbridger, 2020-04-23:
One issue with little router chips (tplink) is that the uarts don't bring out the RTS/CTS handshake lines, which makes serial rather flakey if you are doing any real work with it.
This is needed on one or both of the non-console uarts.

xon/xoff is not especially reliable , measured at 20chars latency in openwrt routers.
(RTS is also needed for RS485 direction)

fifteenhex, 2020-04-23:
The FUART port has RTS/CTS signals. I haven't tried them but the UART is (probably) the well support dw uart.

sbridger, 2020-04-23:
The issue (with routers etc) is: are the pins brought out?
Looks like they are on sch.

BTW I have a serial I2C adaptor unit made for use with little tplink routers. Aside from the main fn, it provides a reliable watchdog and power-down sleep service to the router. OpenWRT on the routers boot pretty fast, so the battery operated units are just turned off for low power. This is also the only absolutely reliable way of regaining control of a locked-up device.

If this is of use to you for a project, you are welcome to one, just ask.

fifteenhex, 2020-04-24:
>The issue (with routers etc) is: are the pins brought out?

They are brought out. I'm not sure if the pinmux code is there to use them yet but I can check.

>BTW I have a serial I2C adaptor unit made

That would be handy. Could you drop me a line via email?