breadbee/breadbee #13: [TODO] Add control for ARM core rail for deep sleep

fifteenhex, 2020-04-24:
pm_gpio4 seems to a magic pin that is used to control the ARM core power rail.

Add a mosfet or something that pm_gpio4 can control to power down the arm core.

Maybe remove pm_gpio4 from the header and put ir_in there instead.

kadamski, 2020-04-24:
Do you have any code reference you could point to which is related to going to sleep?

fifteenhex, 2020-04-25:
The C part is here:
https://github.com/fifteenhex/linux/blob/msc313e_dev_v5_6_rebase/arch/arm/mach-mstar/pm.c

What this basically does is load the assembly part into SRAM and sets a resume address in some registers you'll also see in the boot rom. The boot rom does a check for a magic number and jumps to the resume address back into the kernel.

The assembly part is here:
https://github.com/fifteenhex/linux/blob/msc313e_dev_v5_6_rebase/arch/arm/mach-mstar/suspend-msc313.S

The comments describe what it's doing (mainly ddr self-refresh setup and turning stuff off). The bit I worked out yesterday is right at the bottom. I could never get pm_gpio4 to toggle in my gpio driver.
I worked out that pm_gpio4 is special. The register that is written with 0xbabe locks the pin high so you can't accidently turn the power off from what I can tell.

fifteenhex, 2020-04-25:
Scratch the mosfet bit. Looks like there is something internal that disconnects the power that can be controlled from pm_gpio4.

fifteenhex, 2020-04-26:
Board rework isn't needed. Almost working in software now.

closed 2020-04-26 by fifteenhex