breadbee/breadbee #14: Driver priority and boards?

mbilloo, 2020-04-24:
I'd like to help out with driver work (I have a fair amount of experience working on kernel drivers). What should be priority? Also, do you have any more boards you're giving out/selling?

Thanks

fifteenhex, 2020-04-25:
Right now my priority is to try to get the base machine support into mainline. I want to do this as soon as possible so someone doesn't go and remove the barely used driver for the USB host controller.

Right now I have PCBs left for 4 boards that I'm going to build up and try to send out to people.
I got tons of requests for boards so I might have to be a pretty selfish and send them out to people that would make the most impact before I order the next set of PCBs.

Someone that is really good at thumb assembly that can reverse engineer the DRAM init and add it to the uboot SPL so no blobs are needed to boot is probably the highest priority.

mbilloo, 2020-04-25:
This may be a silly question, but it looks like the bootrom explicitly attempts to load the IPL at a particular address in either SPI NOR or on the SD card. Is the idea that u-boot will simply be at the location of the IPL and so the bootrom will think it's loading IPL but it will instead be loading u-boot (which will do the necessary DRAM setup and rest of the init)?

mbilloo, 2020-04-25:
Also,  where could I find the IPL blob? The linux-chenxing repo doesn't seem to have it.

fifteenhex, 2020-04-25:
> Is the idea that u-boot will simply be at the location of the IPL and so the bootrom will think it's 
> loading IPL but it will instead be loading u-boot

Exactly. At the moment because I haven't fully worked out the DRAM setup it is loading the IPL, then uboot's SPL then uboot for real. If we can do the DRAM setup in the SPL we can get rid of the IPL and everything is opensource. The current uboot source already has an option for building the SPL with the right header for the bootrom to think it's an IPL.

The current IPL is here: https://github.com/breadbee/breadbee_buildroot/blob/master/br2breadbee/blobs/IPL.bin