breadbee/breadbee #15: [TODO] Work out crypto engine

kadamski, 2020-04-26:
Some work is already done in https://github.com/fifteenhex/linux/blob/msc313e/drivers/crypto/msc313-aesdma.c. The comment explains how SHA256 block could be used but there are other things this peripheral can do.

Based on disassembly of the bootROM, I was able to mostly confirm the sha256 generation:
- Each time a new operation is done, a value of 0 is written at the offset 0x0
- The 0x28 and 0x2C stores low/high 16 bits of the buffer address
- The 0x30 and 0x34 stores low/high 16 bits of the buffer size
- The 0x74 is set to 0x40 (block size? number of iterations? **worth investigating**)
- bits 9 and 11 of register 0x20 are set to start the operation, all other bits are set to 0 (**what happes if bit 9 is not 1? What is bit 11?**)
- The 0x3c is polled for status until bit 0 goes high
- The output is taken from 0x40 - 0x5f
- Bit 7 of register 0x20 is cleared at the end

The other computation done in bootROM is using those registers:
- 0x0 is zeroed at the beginning
- 0xA0 is set to 0x3f06 (**no idea what that means**) and then a loop waiting for bit 0 in 0xA4 is run
- The data is piped to some fifo? in the backwards order (from last byte to first) via registers 0x8c and 0x90. The bytes are swapped - 0x8c gets (buf[N] & 0xff) << 8 | buf[N] >> 8 while 0x90 gets (buf[N-1] & 0xff) << 8 | buf[N-1] >> 8. After 0x8c and 0x90 are written with buf[N] and buf[N-1]. they are written with buf[N-2] and buf[N-3] until 256 bytes are written.
- the 0x9C is set to 1 and 0xA4 is being polled for bit(1)
- The value 0xc is written to register 0xc (**no idea what that means**)
- The output is taken from registers 0x94 and 0x98, register 0x88 is an index for the output. So you set the index of the bytes you want to 0x88, then you read two bytes from 0x94 and 0x98, then you increment 0x88 by 2.
- After data is read, there is a poll for bit 0 in 0xA4 again (**no idea why**)
- 0 is written to register 0x80 (**no idea why**)
- bit 0 is cleared in 0xA0 at the end

I don't know what exactly is computed here but the last 32 bytes of this computation (after piping 256 bytes at the end of IPL image) are compared with the sha256 of the IPL image to authenticate it. Can this block be even used for something else than that? I would image such authentication would use some key which is baked into the image and is secred - can some other key be used?

fifteenhex, 2020-04-28:
Working this out would be really useful. Especially in uboot as it could make validating the images being loaded a lot faster.

There is a vendor driver here with register/bit locations -> https://github.com/fifteenhex/linux-ssc325/tree/v4.9.84-sigmastar/drivers/sstar/crypto

I haven't touched in ages. I think I could never get it to actually do anything and assumed the clock was wrong.

kadamski, 2020-04-28:
I will have a look at the driver and I will poke with that when i have a board, for sure, to verify the results are what is expected. Are you able to assign this task to me or its not possible as I'm not part of the team or something?

fifteenhex, 2020-04-28:
Awesome. I manually poked the registers in that area for a long time and got nowhere but I think now more of the clocks etc are mapped out it might be easier to get it going. I've assigned this issue to you but I'll try to take a look too once I have the suspend to ram wrapped up.

On you board: Right now I'm waiting for some flux to build it up. I'm not sure why but amazon here had a week wait on all of the different ones they carried. Looks like it'll turn up Thursday so if all goes well it'll be put together over the weekend.

kadamski, 2020-04-28:
I've heard the shipping (especially the international one) takes ages now so I don't have high hopes of the board arriving to me very soon anyways. Nothing beats the possibility to poke around on the real board to verify my assumptions but I still have plenty to do without it for now. In general the BootROM does not seem to hide any hidden gems, like the serial programming protocol you hoped for but it does allow me to verify some details about how the booting works, how peripherals are organised, etc, so I would like to go even further with the analysis. Then maybe I could compare with other bootroms, do you have a dump from newer designs as well? Or maybe I will move on to IPL..

fifteenhex, 2020-04-28:
I'll shoot you an email with some other bootroms.

mbilloo, 2020-04-28:
@kadamski , I've been looking into REing the IPL as well, but at a fetal state. Do you have any recommendations on a disassembler?

kadamski, 2020-04-28:
@mbilloo I'm quite happy with Ghidra currently

fifteenhex, 2020-04-28:
Please keep in mind that currently the only thing we care about in the IPL is the DRAM init. From what I remember of the last time I booted the uboot SPL as the IPL that was the only missing piece to load uboot proper. Most of the rest of it is probably junk we can do without.

mbilloo, 2020-04-28:
@fifteenhex - understood. On that note, @kadamski, could you share your ghidra settings from your work on the boot ROM? Specifically, the block names and their base addresses. My strategy is to use Ghidra and search for references to addresses with LSBs of 0x1018 and see where that takes me. From the datasheet for the MST786 (which I'm assuming has a register file similar to that of the MSC313), that looks to be the register that enables DRAM, which I imagine the IPL must ultimately do.

fifteenhex, 2020-04-29:
The msb2521 datasheet seems to have the right registers for the MIU (dram controller). They match up what is in the vendor code anyhow.

mbilloo, 2020-04-29:
Did you just set the import settings to "default" (i.e. a base address of 0
and the length to the size of the binary)?

kadamski, 2020-04-29:
> could you share your ghidra settings from your work on the boot ROM

I'm a Ghidra noob. I don't use any specific settings. I wanted to create specific memory map with the peripherals at first but I did not find it useful after all so I abandoned that idea. So I just lookup the peripheral base addresses in the devicetree and register offsets in the device drivers in the Linux repo. @fifteenhex started putting this info on linux-chenxing.org and I plan on helping with that as well.

> a base address of 0

Nope. The IPL binary is loaded by BootROM to the SRAM which is at 0xA0000000.

mbilloo, 2020-04-29:
Can you provide the link to the device tree? It seems like there is no
msc313e.dts file in the linux repo.

kadamski, 2020-04-29:
@mbilloo https://github.com/fifteenhex/linux/blob/msc313e_dev_v5_6_rebase/arch/arm/boot/dts/chenxing-v7.dtsi

fifteenhex, 2020-05-03:
FWIW: I've worked out most of what is needed from the IPL now (enabling the PLLs) and I think I'll be able to get the DDR working in the next few days. So don't spend too much time disassembling it.

mbilloo, 2020-05-03:
Nice! Anything else you'd like me to work on?

mbilloo, 2020-05-03:
Also, could you share your strategy in figuring this out. Mine is/was to
basically hunt for all functions in the disassembly that write to the MIU
registers and create a call trace. Of course, I would try to write a script
that would do this for me. There was an interesting thing I did notice when
just looking at some of the functions (I just did a text search in ghidra
for anything that started with 1f2022 initially). For example, there were
writes to some of the internal test registers, which I imagine can just be
copied into uboot.

fifteenhex, 2020-05-04:
>Also, could you share your strategy in figuring this out. 

Mostly what you were thinking. As I know what the MIU registers look like I could find the writes to certain ones to find the right functions. I was mostly looking for the DRAM type and banks/cols being set so I could work out how it knows what memory is in the chip. For DDR2 init there is a single function that ghidra decompiled that can configure for either 32 or 64 megs of DDR2.

Yesterday I worked back through the IPL code until that point and fixed up some things I was missing. For example the serial stopped working as soon as I changed it's clock source from the 12MHz clock that is always on to one of the PLL clocks.. that meant the PLLs weren't turned on. So I worked back through the registers until I found some that looked like they were in the right places and messed around until the uart worked from the PLL clock. The address those registers are in don't seem to be documented in any of the SDKs I can find.

The other interesting bits like setting up the CPU clock come after the DDR is setup so I should be able to get u-boot working without the IPL and then fix up anything that has since stopped working because the IPL isn't doing the setup anymore.

mbilloo, 2020-05-04:
Excellent! Is there anything you'd like me to work on?

Thanks

fifteenhex, 2020-05-23:
@kadamski 

If you're still planning to work on this can you add a comment here https://github.com/breadbee/breadbee/issues/25 ?

Thanks

kadamski, 2020-05-24:
@fifteenhex I do plan on working on that in next week when my day-work related stuff goes a little calmer. I will first try on the dash cam, however, I guess I can expect it to have the same peripheral, right?

fifteenhex, 2020-05-24:
Great. :)
I'm pretty sure the aesdma block is the same (registers, location, interrupts etc) but I'll check.

fifteenhex, 2020-08-01:
The HW RNG is now supported.