breadbee/breadbee #16: [QUESTION] Some UART related questions

kadamski, 2020-04-27:
The UART seems to be mostly compatible with DesignWare UART. The best document explaining register map of this peripheral I was able to find is the https://linux-sunxi.org/images/d/d2/Dw_apb_uart_db.pdf, do you have anything better?

Based on this document, I cannot really understand what is happening in the BootROM so either I have wrong document or..

The code seems to be, for example, writing to register 0x18 which should be MSR which is read-only? What is 0x20 and 0x38? The code also writes 0x14 to 0x00 which is supposed to be THR but this is not any sensible character.. Would that be a pattern to detect the baudrate?

fifteenhex, 2020-04-28:
As far as I can tell the UART is a DW one with one of the registers moved. I added a fixup for the register location to the linux driver.

I think the issue with the addresses you are seeing are because the registers are shifted because of how they are connected to the bus. The shift is 3. So the register offsets are 0, 0x8, 0x18,...

kadamski, 2020-04-28:
Oh, I have to take a look at this fixup, then. And the shift is probably what I was missing. Those tty drivers in the kernel are always hard for me to wrap my head around as there is so much abstraction there :)

fifteenhex, 2020-04-28:
Another thing to look out for is that the register locations are always 2 bytes 4 bytes apart for stuff that's connected via the "RIU". So with blocks like the ethernet and usb that are existing designs and have 32bit registers those registers are split over 8 bytes. It's pretty strange but I think it's maybe because they used the same designs in their TV chips that are driven by some sort of 16 bit 8051 thingy.

kadamski, 2020-04-28:
Yes, generally there is a lot of registers which internally only have 16 bits but take 32 bits. And even though they could be packed into one 32bit register, they are split into H and L 16bit register. Like the peripheral was designed for 16bit CPU.

But back to the UART, I feel a little stupid I did not notice that. I assumed the shift was 4, not 8. When applying shift=8, not all of the registers does make sense. I can see the code setting divisor to 320 which for 12Mhz clock gives nice 38400 baud rate as expected. I can see the 8N1 mode being set, fifos being enabled and reset... all that stuff. The only unknown register is at offset 0x80 (so its is 0x14 before shift). Least significant bit of this register is zeroed as the first operation of the function configuring UART. Any idea what could that be?

Also, as a last action, an yet another unknown pinctrl register (offset 0x14c) is written value 0x3012. Again, no idea what this does but appartenly it must be something related to UART.

fifteenhex, 2020-04-28:
>The only unknown register is at offset 0x70 (so its is 14 before shift). 
>Least significant bit of this register is zeroed as the first operation of the function configuring UART. >Any idea what could that be?

That seems to be after the standard registers. I'm guessing it's something to do with the DMA feature.

>Also, as a last action, an yet another unknown pinctrl register (offset 0x14c)
>is written value 0x3012. Again, no idea what this does but appartenly it must 
>be something related to UART.

0x14c is called "uart sel". It seems to be for selecting which UART is behind the pins. It seems like it's even possible to connect the pm mcu (if there really is one..) to the UART as well.

```
//#define REG_UART_MODE          0x1F203C0C
//#define REG_UART_SEL           0x1F203D4C
//#define REG_UART_SEL4          0x1F203D50
#define REG_FORCE_RX_DISABLE   0x1F203D5C
```

closed 2020-05-03 by fifteenhex