breadbee/breadbee #19: [TODO] Do DDR init from uboot SPL

fifteenhex, 2020-05-02:


fifteenhex, 2020-05-10:
```
U-Boot SPL 2019.07-00041-g697703c390-dirty (May 10 2020 - 22:)

cpuid: 410fc075, mstar chipid: d9
normal power on
wrote 1f206700 <- 2201 was 8801, readback 2201
wrote 1f206704 <- 0420 was 0420, readback 0420
wrote 1f206708 <- 0041 was 0000, readback 0041
wrote 1f20670c <- 0000 was 0000, readback 0000
wrote 1f206720 <- dd2f was 5544, readback dd2f
wrote 1f206724 <- 0024 was 0024, readback 0024
wrote 1f20672c <- 0000 was 0000, readback 0000
wrote 1f206728 <- 0001 was 0001, readback 0001
trying to work out DDR config..
mystery gpio register is 01
Detected DRAM: 04000000 bytes
wrote 1f206208 <- 0100 was 0100, readback 0100
wrote 1f20620c <- 0216 was 0218, readback 0216
wrote 1f206210 <- 0010 was 8010, readback 0010
wrote 1f206204 <- 0000 was 0100, readback 0000
delaying for 1000
-- 0 --
wrote 1f20243c <- 0c01 was 8004, readback 0001
wrote 1f202c0c <- fffe was 0000, readback fffe
wrote 1f202c8c <- ffff was 0000, readback ffff
wrote 1f202d0c <- ffff was 0000, readback ffff
wrote 1f202d8c <- ffff was 0000, readback ffff
wrote 1f202e0c <- ffff was 0000, readback ffff
wrote 1f202e8c <- ffff was 0000, readback ffff
wrote 1f202f0c <- ffff was 0000, readback ffff
wrote 1f2023cc <- fffe was 0000, readback fffe
wrote 1f2020f0 <- 0001 was 0000, readback 0001
delaying for 1000
wrote 1f202048 <- 1000 was 0000, readback 0000
delaying for 1000
wrote 1f202048 <- 0000 was 0000, readback 0000
delaying for 1000
-- 1 --
wrote 1f20206c <- 0400 was cc00, readback 0400
wrote 1f202068 <- 2004 was 0000, readback 2004
wrote 1f202114 <- 0001 was 0301, readback 0001
wrote 1f202060 <- 8000 was 0200, readback 8000
wrote 1f202064 <- 0029 was f000, readback 0029
delaying for 100
wrote 1f202114 <- 0001 was 0001, readback 0001
wrote 1f2020bc <- 0010 was 0000, readback 0010
wrote 1f202044 <- 0004 was 0000, readback 0004
delaying for 100
wrote 1f202058 <- 0114 was 0014, readback 0114
pll status before: 0000 after: 0000
-- 2 --
wrote 1f202404 <- 0392 was f000, readback 0392
wrote 1f202408 <- 000d was 0c09, readback 000d
wrote 1f20240c <- 1828 was 0008, readback 1828
wrote 1f202420 <- 0003 was 0000, readback 0003
wrote 1f202424 <- 4004 was 4000, readback 4004
wrote 1f202428 <- 8000 was 8000, readback 8000
wrote 1f20242c <- c000 was c000, readback c000
wrote 1f202050 <- 0020 was 8000, readback 0020
wrote 1f2025a4 <- 6000 was 0000, readback 6000
wrote 1f202644 <- 0003 was 0000, readback 0003
wrote 1f20267c <- 0000 was 0000, readback 0000
wrote 1f202680 <- 0909 was 0303, readback 0909
wrote 1f202684 <- 071e was 0208, readback 071e
wrote 1f202688 <- 2707 was 0c01, readback 2707
wrote 1f20268c <- 0908 was 0601, readback 0908
wrote 1f202690 <- 0905 was 0603, readback 0905
wrote 1f202694 <- 0304 was 0301, readback 0304
wrote 1f202698 <- 0528 was 0528, readback 0528
wrote 1f20269c <- 0046 was 000e, readback 0046
[... 208 lines cut ...]
1f202760: 0000 0000 0000 0000 0000 0000 0000 0000 
1f202770: 0000 0000 0000 0000 0000 0000 0000 0000 
1f202780: 0000 0000 0000 0000 0000 0000 0000 0000 
1f202790: 0000 0000 0000 0000 0000 0000 0000 0000 
1f2027a0: 000e 0000 0020 0000 40c3 0000 0000 0000 
1f2027b0: 0c80 0000 0000 0000 0000 0000 8000 0000 
1f2027c0: 0000 0000 0000 0000 0000 0000 0000 0000 
1f2027d0: 0000 0000 0000 0000 0000 0000 0000 0000 
1f2027e0: 0000 0000 0000 0000 0000 0000 0000 0000 
1f2027f0: 0001 0000 0000 0000 0000 0000 0000 0000 
testing ddr..
wrote aaaa5555 to 20000000, was 0000b333, readback aaaa5555
wrote a5a5a5a5 to 20000004, was 0000c831, readback a5a5a5a5
efuse_20: 02dc, utmi0_50 0000, utmi1_50 0000
Trying to boot from SPI
unrecognized JEDEC id bytes: 00, 00, 00
SPI probe failed.
Trying to boot from UART
CCCCCCCCCCspl: ymodem err - Timed out
SPL: Unsupported Boot Device!
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###
```

Working for M5 now.

fifteenhex, 2020-05-10:
```
wrote aaaa5555 to 20000000, was 00000002, readback aaaa5555                           
wrote a5a5a5a5 to 20000004, was 00005049, readback a5a5a5a5                           
efuse_20: 02dc, utmi0_50 0000, utmi1_50 0000                                          
Trying to boot from SPI                                                               
unrecognized JEDEC id bytes: 00, 00, 00                                               
SPI probe failed.
Trying to boot from UART                                                              
C CRC mode, 1956(SOH)/0(STX)/0(CAN) packets, 4 retries
Loaded 250036 bytes
!

U-Boot 2019.07-00042-gc9b357418b-dirty (May 10 2020 - 22:52:09 +0900)

DRAM:  64 MiB
WARNING: Caches not enabled
Loading Environment from SPI Flash... unrecognized JEDEC id bytes: 00, 00, 00
*** Warning - spi_flash_probe_bus_cs() failed, using default environment

In:    uart@1f221000
Out:   uart@1f221000
Err:   uart@1f221000
Net:   No ethernet found.
Autoboot in 10 seconds
=> sf probe
unrecognized JEDEC id bytes: 20, 40, 16
Failed to initialize SPI flash at 0:0 (error -2)
=> 
```

Uboot even works :)

fifteenhex, 2020-05-12:
With IPL

```
=> dhry 
1000000 iterations in 717 ms: 1394700/s, 793 DMIPS
```

With SPL

```
=> dhry 
1000000 iterations in 26298 ms: 38025/s, 21 DMIPS
```

:-1:

kadamski, 2020-05-13:
@fifteenhex details ;)

fifteenhex, 2020-05-13:
It's not too bad once it's booted by extracting the kernel takes about 4 minutes at the moment.
I think the CPU is running from the 12MHz clock but I can't work out what in the IPL is switching it.

kadamski, 2020-05-13:
But once it is booted it works fast so someone is switching this to correct clock, right? And it isn't u-boot.. so must be in the kernel code, correct?

fifteenhex, 2020-05-13:
I don't think the cpu is running faster once booted. The environment is light enough that after the big initial XZ decompress for the kernel it feels ok. Some naive tests like gzipping a 1MB of random data show it's still pretty slow. Somewhere around 10 times slower after booting with the SPL.

I'm pretty sure there is a bit to switch the cpu clock between the 12MHz always on clock and the special cpu pll somewhere in the pm area but I'm not spotting anything that stands out in any of the IPLs I have. The cpu pll seems to lock correctly when I trigger it to change frequency so it's at least partially working.

Anyhow much closer to having a fully open source setup now.

fifteenhex, 2020-05-14:
```
=> dhry             
1000000 iterations in 48834 ms: 20477/s, 11 DMIPS
=> md.w 0x1f2041f0 1
1f2041f0: 0000                                       ..
=> mw.w 0x1f2041f0 1
=> dhry             
1000000 iterations in 440 ms: 2272727/s, 1293 DMIPS
```

kadamski, 2020-05-14:
Nice! The same on M3 - 793 DMIPS vs 11 DMIPS when 0x1f2041f0 is != 0

fifteenhex, 2020-05-14:
With the SPL it's a bit different:

```
=> dhry 
1000000 iterations in 25733 ms: 38860/s, 22 DMIPS
=> mw.w 0x1f2041f0 0
=> dhry             
1000000 iterations in 218087 ms: 4585/s, 2 DMIPS
=> 
```
At least proves that register is switching between clocks though.
I'm guessing the processor PLL is wired to the wrong source or one of the variables is wrong as if I change the values to the ones I have have 400MHz, 800MHz and 1GHz I can see the DMIPs number going from 19, 20 and 22 between the different speeds.

kadamski, 2020-05-14:
What settings are you changing to switch between 400/800/1000 Mhz mode?

fifteenhex, 2020-05-14:
I'm doing the same thing as I'm doing in the kernel for cpufreq.

https://github.com/breadbee/u-boot/blob/27d4c99c85b278cae8361cd9c36413c456f29519/board/thingyjp/breadbee/clk.c#L73

fifteenhex, 2020-05-19:
FWIW: There is a bunch of registers in the gpio area that are used to somehow measure the current clock. The read out there is the same for the IPL and the SPL so the cpu PLL seems to be running at the right frequency. Very weird.

fifteenhex, 2020-05-24:
dcache setup is wrong somewhere it seems. the msc313e shows the same dhry results in uboot when dcache isn't enabled.

fifteenhex, 2020-05-31:
Fixed. SMP bit wasn't getting enabled so the caches weren't being used.
Porting to msc313e now.