breadbee/breadbee #20: [QUESTION] Power related questions

kadamski, 2020-05-04:
I've noticed that the MSC313e data brief mentions 0.9V as the typical Core Power Supply Voltage  (while 0.87 is minimal, so quite close). You are using 1.0V, however, which seems to be fine since it is much lower than the absolute maximum (1.26V) and the timing diagrams in the brief also mentions 1.0V/0.9V as VDD. So I was wondering - did you choose 1.0V just to be on a safe side? Maybe this is the reason why, even though the CPU is rated to 800Mhz, you can easily overclock it to 1GHz? Maybe with 0.9V that wouldn't be possible?

Also, I wanted to understand the role of R21, R22 and R18. Why aren't the two resistors creating a  voltage divider enough there?

fifteenhex, 2020-05-05:
I went for 1v as I didn't have a reference schematic and that's what I measured on an existing product. I think I have an cpufreq table with the *right* core voltages for different frequencies now.

The E version of the part runs fine a 1GHz. 1.2GHz works as well but other parts of the chip don't like it and eventually the CPU gets stuck on a bus access from what I remember. The older non-E version only runs at 800MHz and gets stuck when it tries to ramp up to 1GHz even with 1V core supply.

R21, R22 and R18 were intended to add extra load if the load presented by the soc wasn't enough for the ea3036c as I had issues with that with the first 2 layer incarnation. The 100k value doesn't mean much. I'll probably get rid of them at some point to make space.

kadamski, 2020-05-05:
So the SoC has capability to change the core voltage to get smaller power consumption / heat generation internally based on this 1V input voltage? Meaning the actual value does not matter that much..

fifteenhex, 2020-05-05:
The regulator needs change the core voltage to save power. I think that could be done with a resistor and one of the gpios to change the feedback for the core voltage buck converter.

kadamski, 2020-05-05:
But there is not much to be gained if we cannot really go below 0.9V, right?

fifteenhex, 2020-05-05:
I don't think so. In sleep with the memory in self refresh is seems it'll be around 25mA.
To get much less I think the DDR would need to be turned off completely and that means adding a real PMIC really.

kadamski, 2020-05-05:
The DDR in self refresh mode could be powered differently than in normal mode?

fifteenhex, 2020-05-06:
mm not sure about that. Without knowing what DDR chip is actually in there it's hard to know if it'll like the DDR voltage changing or not.

closed 2020-05-19 by fifteenhex