breadbee/breadbee #21: MUSB USB device controller?

mbilloo, 2020-05-05:
@fifteenhex since @kadamski is working on the crypto engine, I thought I could tackle the USB controller, unless you had something else with higher priority?

fifteenhex, 2020-05-10:
Sorry for the delay.

Getting the usb host controller working would be awesome. The only thing I can think of right now aside from that that I really want is an SD controller driver in u-boot.

You'll need some hardware to work on that I think. Right now I can't get hold of some of the parts I need to build up more boards but I can buy you a dash camera that has another chip in there that is almost identical for you to hack on. I've been using one for a lot of the reverse engineering as it's a bit easier to unbrick.

mbilloo, 2020-05-10:
That would be awesome. Just send me the link and I can order it myself
(that may be quicker).

fifteenhex, 2020-05-11:
The 70mai lite is the best thing I can find at the moment. If you crack it open you get easy access to the serial port and a button to trigger booting from SD. Also the usb charge connector has the USB data singles wired to it so it should be usable for testing the usb host controller.

It's about $50:
https://www.gearbest.com/car-dvr/pp_3006867820113255.html

mbilloo, 2020-05-13:
I ordered the dash cam and am waiting for it to come in. Could you send a photo outlining the location of the serial port ? I imagine that it's TTL output, correct?

Thanks

fifteenhex, 2020-05-13:
I don't have mine to hand at the moment otherwise I'd send a photo but if you look here: https://fccid.io/2AOK9-MIDRIVED08/Internal-Photos/internal-photos-4351132.pdf
The page labelled 5 of 8 has a photo of the guts. The pads between the two push switches on the bottom corner are the UART. GND RX TX 3.3V from left to right if I'm remembering correctly.
The button on the left side forces it to boot from SD card.
It's 3.3V logic so you'll need a USB to serial that uses 3.3v levels like the cp2102 ones.

kadamski, 2020-05-14:
[image]

TX RX GND 3.3V

mbilloo, 2020-05-19:
Since it looks like the SD card driver works on the mercury5 (since I can boot off of the SD card), do you want me to move to the USB controller kernel driver?

fifteenhex, 2020-05-19:
At the moment u-boot itself can't load from SD. In the rtk file u-boot and the kernel are joined together to trick the IPL into loading both into memory.

Once I have the u-boot SPL working properly an SD driver will be needed to load the full u-boot as the SPL will be running the show. I plan on porting the kernel driver I wrote to u-boot at some point. If you wanted to take a look at that before I get to it I wouldn't stop you... ;)

The USB controller is probably more interesting though. I think it'll just need the musb glue bit implementing properly and maybe some changes to the phy/mux part to connect the port to the controller. I started on it here: https://github.com/fifteenhex/linux/commit/6f5a574163a42ce7617c846ea981f269c49c5449
The registers and bit meanings for the MUSB controller are listed in the msb2521.pdf datasheet so there is something to actually work from.

Once that works you could make the mercury5 camera into a usb ethernet gadget and getting stuff in and out of it would be a bit easier.

fifteenhex, 2020-05-22:
FYI: u-boot sd driver is in progress.

mbilloo, 2020-05-22:
Great! I'm balancing my day-time job and this effort. I'll keep you posted
as I make progress on the USB controller driver.

fifteenhex, 2020-05-23:
Please feel free to work at your leisure. Also please drop me a line if you want to be reimbursed for the cost of the camera. Also please leave a comment here https://github.com/breadbee/breadbee/issues/25 so I can keep track of allocating dev boards. :)

Anyhow, the driver is pretty rough but u-boot can see sd cards now:

```
field write, off 30, mask 2, shift 1, value 0, masked value 0
field write, off 30, mask 1, shift 0, value 0, masked value 0
field write, off 30, mask 100, shift 8, value 0, masked value 0
                MMC_RSP_R1,5,6,7         0x00000920 
=> mmc info
mstar_mmc_getcd:492
Device: MSC
Manufacturer ID: 5d
OEM: 5342
Name: N1BN1 
Bus Speed: 25000000
Mode: SD Legacy
Rd Block Len: 512
SD version 1.0
High Capacity: No
Capacity: 1.9 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
```
I think this will make life a lot easier working on the camera. I'll add some notes on using it to the breadbee_dev page at some point.

mbilloo, 2020-06-06:
"musb_am335x.c" referenced in https://github.com/fifteenhex/linux/commit/6f5a574163a42ce7617c846ea981f269c49c5449 doesn't exist. https://github.com/fifteenhex/linux/commit/4baa550ecc86693106493bd92382e0edb8caf64d says to drop the dummy driver, but it looks like your commit added the inclusion of the driver back in, but not the source itself. What am I missing?

fifteenhex, 2020-06-07:
That's going to be a rebase fail on my part. I probably merged those lines back in when they got dropped from mainline and a I rebased. You can take them out.