breadbee/breadbee #22: Flashing u-boot?

mbilloo, 2020-05-17:
Got the dashcam, connected UART, and didn't blow anything up. Are there instructions on flashing uboot?

mbilloo, 2020-05-17:
It looks like instructions are in the breadbee_dev and buildroot repos. However, when I boot, I don't have the standard u-boot shell, it looks like something else. How can I get to u-boot?

Thanks

fifteenhex, 2020-05-18:
Sorry, for some reason I didn't get a notification.

With the mercury5 chips booting directly from SD works. On that camera you can disconnect the battery and then press the button closest to the battery connect while plugging in the usb to get it to boot from SD. I've made an SD card with the right format with u-boot and a kernel here: https://drive.google.com/file/d/1xbFF7_SD8Ws4wTvUJUDp8DqNqJAbw0-C/view

Some notes on how to boot the kernel there are here: https://github.com/breadbee/breadbee_dev#booting-the-uboot--kernel-fit-rtk-image

I'll try to put together some better documentation for how to do u-boot/kernel hacking using the camera.

mbilloo, 2020-05-18:
Do you want me to do that as I progress? I can update the README to point to the appropriate links as I move along and submit a PR once I'm done. This way, if there are any gaps in the documentation, I can do it right away instead of you having to search through this exchanges later on.

fifteenhex, 2020-05-19:
I'll clean up the dev repo a little bit and we can go from there. The plan was to make it possible to check that out, run a script and start hacking but I never got around to it. I'll try to fix that in the next few hours.

fifteenhex, 2020-05-19:
I've updated the breadbee_dev repo with just enough to build the "rtk" image for doing u-boot/kernel work on the mercury5.

My workflow at the moment is to tweak stuff in the kernel or u-boot, build the rtk and copy it to SD card. It's not as nice as booting from tftp.. usb ethernet does work on the m5 once the kernel is loaded so it's possible to replace the rtk file from linux if that makes things quicker.

mbilloo, 2020-06-06:
@fifteenhex - cool! Instructions work. 

"usb ethernet does work on the m5 once the kernel is loaded..." - Doesn't the driver for the USB controller need to be working to support USB ethernet?

mbilloo, 2020-06-06:
Never mind, it looks like the driver already supports host mode (hence the support for Ethernet over USB), but not in device mode, hence the issue. I'll move this conversation over to that issue.

closed 2020-06-06 by mbilloo