breadbee/breadbee #23: [TODO] Pay someone to do "production ready" layout fixes.

fifteenhex, 2020-05-19:


bencorrado, 2020-06-25:
Do you have a list of asks here or just have someone review then make changes based on the review.  I could potentially help here and would not be looking for a few.

fifteenhex, 2020-06-25:
A review by a proper EE or just someone that knows what they are doing would be great. I just tinker so going ahead and trying to make 2000+ of these scares me a little. ;)
Things that I know are certainly wrong are things like the routing of the USB signals. I didn't think I'd ever get the software side of the USB working so I just crammed them in instead of trying to route them as a proper pair.

kadamski, 2020-06-25:
You mean the USB signals on the 1mm header? Fortunately, the board is small itself so the effect of those traces is not that big. The bandwidth of the USB signal is ~90Mhz which means a wavelength of ~170cm. So the 1/10 of that is 17cm which is way more than the 30mm the dp/dm signals on the breadbee have. The two lines are not matched in size but the skew is 30.385 - 28.869 = 1.5 mm out of 45mm allowed (this includes the skew in cable as well, of course but still there is a large margin). Kicad do have an easy way to control match the length of the two wire pairs so this could be easily improved. The biggest problem is, probably, the fact that you are going over vias and especially dp has this sharp angle. Also the fact that headers are used is not without effect, I guess.

The only other thing to think of is the impedance of the pair which should be 90 ohm. This is controled by the track width and depend on the spacing between lines (6 mils in your case?), the material of between the layer the trace is going and the first inner layer and the high of the dielectric between them. JLCPCB has a nice calculator you can use to calculate the width of the trace for their 4 layer boards on https://cart.jlcpcb.com/impedanceCalculation. So for example for their normal board (1.6mm thick) you should be using 9 mils (0.22mm) trace width wich is a little more than you have now (for the current 0.15mm width you will get ~105 ohm of differential impedance with 6 mils spacing) or you would have to reduce the spacing between traces to 4 mils which would give you ~95 ohms. 

Of cource the problem of calculating the impedance is that it depends on the dielectric that is used between the layers (the type and the height) so you would first need to decide where you are going to manufacture the boards (we know it's not in JLCPCB, right? ;)).

The good document describing this is https://www.silabs.com/documents/public/application-notes/an0046-efm32-usb-hardware-design-guidelines.pdf. But, well, as you said, a look from a proper EE would probably be even better :)

kadamski, 2020-06-25:
Oh, I forgot to check the actual inner layer adjacent to the traces (In2.Cu) - it is not solid (you trace some power lines there :/) and there are a lot of voids in it above the wires which is bad as it dos not make a good return path for signal. This may be the biggest problem.

The oscillator is far away so this should not be a problem but the traces are going very near of the inductors used by ea3036 which is a switching mode regulator and this should be avoided. Also they are going near SPI traces which might be considered high speed signals as well (what is the max frequency of those SPI lines - 25Mhz?) so it would best be avoided too.