breadbee/breadbee #24: [QUESTION] Boot from SDIO / SD CARD Interface

kjngineering, 2020-05-21:
Hi fifteenhex,

Looks like a really interesting mcu - I might actually have some application for me. The dual SPI busses makes it very interesting!

Based on your current understanding of the chip, do you think it is possible to have it jump to a SPL that is in a region of an SD Card (ala Allwinner) as opposed to SPI NOR flash? 

Assuming this is possible would this allow you access two the two SPI busses for devices?

Thanks

fifteenhex, 2020-05-22:
From what I can tell the bootrom can't boot from the SD controller that is broken out on the msc313e chip so you have to have something on SPI NOR to boot.
I intend on porting the kernel driver for the SD controller to u-boot so u-boots SPL can load everything else from SD which would mean you could probably use an 8 megabit/1 megabyte flash.

If you just need two SPI buses then you can already do that as there are two SPI controllers aside from the memory mapped one for SPI NOR.

fifteenhex, 2020-05-24:
FYI: The uboot port now has some clunky sd support that is enough to load the kernel from SD.
You still need some SPI-NOR but 1MB is enough for uboot.

closed 2020-11-29 by fifteenhex