breadbee/breadbee #25: [XXX] Dev board requests

fifteenhex, 2020-05-22:


mbilloo, 2020-05-23:
@fifteenhex, I'd like to request a board.

fifteenhex, 2020-06-05:
[image]

Work in progress.

1 fully built and packaged for @kadamski 
1 built but waiting for 1.0mm pitch header for @mbilloo 
1 built but waiting for 1.0mm pitch header for <next person that wants one>
1 waiting for usb connector, 1.0m pitch header
8 pcbs left

brentpicasso, 2020-06-05:
I would be interested in a board and can help with testing. I have a small batch of processors that arrived from ebay, if you're short on processors I can solder my own. Thanks!

afaerber, 2020-06-05:
I would be interested in testing kernel patches and the boot mechanism with a view to [EBBR](https://github.com/ARM-software/ebbr) compliance, for booting distros such as openSUSE Tumbleweed. But I'm currently probably too busy to contribute much patches, so consider me lower priority. Thanks.

C-o-r-E, 2020-06-05:
@fifteenhex interested in getting netboot going if not already working. I just received some MSC313E from ebay so in theory i could build some too. Happy to be in the lower priority gang with @afaerber

fifteenhex, 2020-06-06:
@kadamski @mbilloo @brentpicasso @afaerber @C-o-r-E

Can you all leave a note about which 2.54mm header you want. If you want to use the breadboard adapter a right angle header needs to be fitted.

Thanks.

ConnorRigby, 2020-06-06:
Interested in a board to get a buildroot port up and running. Specifically for [nerves](https://nerves-project.org).

fifteenhex, 2020-06-06:
> Edit: just reread the readme and found that there is already buildroot.

Taking some of the config etc that's in the buildroot repo and trying to get it into buildroot would be useful. Do you want to mount it in a breadboard?

kadamski, 2020-06-06:
@fifteenhex You can leave the 2.54 header unpopulated. I'll solder one if I find it useful. Thank you.

mbilloo, 2020-06-06:
@fifteenhex , same with me. I can solder on headers. Thanks!

C-o-r-E, 2020-06-06:
@fifteenhex 90 degree headers would be nice.

brentpicasso, 2020-06-06:
you can skip headers for mine. Thank you!

fifteenhex, 2020-06-06:
@kadamski 
That's lucky.. I tried to send you one with headers. :)

For people that don't want headers do you just not want the 2.54mm one? I'm waiting on the 1.0mm pitch headers to come.

kadamski, 2020-06-06:
@fifteenhex If it has headers, that's ok too. I just don't have any preference:)

brentpicasso, 2020-06-06:
No headers at all - I'll custom wire things if I need to.  Thanks!

mbilloo, 2020-06-06:
If it has headers that's fine. I'd rather have the board sooner. Thanks

afaerber, 2020-06-06:
@fifteenhex Right angle please - unless there's any disadvantage to straight? Like, were you considering to design any expansion boards that would require both 1 mm and 2.54 mm connectors facing the same direction? (thinking of 96boards mezzanines with LS and HS connectors)

@C-o-r-E PXE boot from U-Boot I was also hoping for, to speed up turnaround times for kernel development. However, the [Readme](https://github.com/breadbee/breadbee#u-boot) only mentions SPI-NOR (and not sure what "boot process below" refers to there, it may have moved to http://linux-chenxing.org/blobs/ipl.html?).

I'd like to offer my help in getting the OpenOCD config upstream - we'll need to split it into three files with some conditionals (and add @fifteenhex's copyright), then we can push SoC and board configs to Gerrit.

fifteenhex, 2020-06-07:
@afaerber 

On the headers:

My feeling with the 2.54mm header for the production boards is that it comes unpopulated and you solder whatever works for your project. All the boards I've built either have a right angle sticking out the bottom for breadboarding, or vertical on the ethernet jack side but I could imagine someone wanting it vertical sticking out of the opposite side so the board can be mounted in parallel on top of something.

For reference is this the general idea when using the right angle connector:

[image]

[image]

[image]

On booting:

Booting via TFTP from u-boot works. You always need SPI-NOR as it doesn't seem possible for the boot rom to load from anything else but once u-boot is running ethernet works, mmc sort of works, usb could potentially work as there is a driver for the usb host controller. Also recently I have worked out how the memory init works so having a completely open source path from the boot rom is possible now.

mbilloo, 2020-06-07:
@fifteenhex , what's the ETA on you shipping boards out? I ripped one of the UART pads off my dashcam and am debating of getting another one off ebay. If you'll be shipping in the next few days, I'll hold off.

fifteenhex, 2020-06-07:
Ouch. Instead of ordering a new one I would first try to track the trace back and see if there is a via you can scratch off and tap on to. I would find one on my camera for you but that area on mine is covered with hot glue to stop the wire yanking the pads off.

ETA on shipping... I have one board that is fully complete but with vertical header. All of the others are waiting on 1.0mm headers to come from China. Headers should come next week.
There's a bit of an issue with shipping at the moment because the post office here has stopped shipping to some countries. I'm trying to work out an alternative at the moment.

mbilloo, 2020-06-07:
Lol, too late. Just ripped off a few more pads while looking for vias (FYI, it looks like there are no obvious vias.  The traces for Vcc, Gnd, Rx, Tx go right underneath the big black square with the shield underneath). 

My problem was that I soldered the headers directly on the pads instead of having wires and hot gluing them down, which wasn't the smartest thing to do. C'est la vie. Another cam is on the way soon.

fifteenhex, 2020-06-12:
If possible can everyone leave a note of what country they are in? I need to work out how to ship boards. At the moment some places are looking like $100+ :(

brentpicasso, 2020-06-12:
USA. feel free to send it in the equivalent of a first class envelope / padded envelope.  Thanks!

mbilloo, 2020-06-12:
USA

kadamski, 2020-06-12:
@fifteenhex: Poland. If the shipping is expensive, just skip me (at least until some cheap option becomes available again). I've seen that the [latest status](https://www.post.japanpost.jp/int/information/2020/0604_01.html) is from June 4th and Poland is still on the list (just like most countries in North America and Europe, I think :/).

fifteenhex, 2020-06-12:
@kadamski From what I can tell the local courier (yamato) has an international service that is still running. They say "delayed" for US and UK parcels but not totalled stopped. If I can get your board to the UK at least I can have someone send it on from there.

C-o-r-E, 2020-06-13:
@fifteenhex Canada here. Good to hear about TFTP working. I like the idea of minimizing the need for the NOR flash as much as possible.

fifteenhex, 2020-06-13:
> @fifteenhex I like the idea of minimizing the need for the NOR flash as much as possible.

Are you thinking of putting the root fs on NFS? You can use an initramfs and have everything in memory but with only 64MB to work with I think that gets pretty tight quickly. Last time I used the in-kernel NFS root support it was pretty hard to get working correctly..

With the NOR flash it has a controller that memory maps it so it's pretty quick and the work to move data from NOR into main memory can be DMA driven so it takes no CPU effort.

C-o-r-E, 2020-06-13:
I'm thinking that for development workflow its best to be pulling as much as possible over the network. What is your current workflow for flashing the NOR flash?
(maybe this topic should move to another thread or medium)

fifteenhex, 2020-06-14:
> What is your current workflow for flashing the NOR flash?

My current kernel/u-boot dev flow is here: https://github.com/breadbee/breadbee_dev
I'm mostly using TFTP to load new kernels and using the existing rootfs in nor flash.
The user focused buildroot based flow also comes with a tftp server built in so the user
can make their changes and push them via tftp without installing anything extra on their
system.

fifteenhex, 2020-06-21:
The 1.0mm connectors came and all of the boards except one are now built.

There seems to be an issue with the usb connectors I've gotten recently as they don't wet with the solder well and pop off if you're a bit rough on them.. so all of these boards have some epoxy behind the connector to make it harder to pull it off. But either way please be careful....

Anyhow I will start shipping them tomorrow. If you haven't sent me your shipping address via email yet please do.

mbilloo, 2020-07-07:
@fifteenhex , just wanted to confirm that you received my shipping address via email? Thanks

fifteenhex, 2020-07-08:
> @fifteenhex , just wanted to confirm that you received my shipping address via email?

Yes. Boards will be going out soon. It's been difficult to find time to go to the post office in between work, getting sick and now the crazy weather going on here. I'll send you the tracking number once I have one.

mbilloo, 2020-08-11:
@fifteenhex , just checking in again on the status of the boards. Thanks!

fifteenhex, 2020-08-11:
> @fifteenhex , just checking in again on the status of the boards. Thanks!

Your board is packaged and sitting on my desk with the others. The only one I've managed to get the post office to take so far was the one for @kadamski .
Airmail is still suspended to a lot of places: https://www.post.japanpost.jp/int/information/2020/0806_01_01_en.pdf

Once stuff starts moving again I'll start sending them out.

rperier, 2021-04-22:
Hi, I would be interested in writing kernel mainline patches  (I am active upstream on severals platforms for years). I am also a debian maintainer, I could help for testing and add support in debian.

Regards,

fifteenhex, 2021-04-23:
> Hi, I would be interested in writing kernel mainline patches (I am active upstream on severals platforms for years).

Taking some of the current backlog of ~300 patches would really help.
For example the clocks need to be totally reworked because mainline will not accept the way I implement it..

>I am also a debian maintainer, I could help for testing and add support in debian.

Debian is probably impossible in 64MB of RAM right? The SSD202D with dual A7 and 128MB of RAM would probably be more suitable there. They are almost the same internally so anything that helps the SSD202D helps this board and all of the other chips in this family.

Anyhow it seems you are in France? I have a bunch of boards made up for the people in this thread but even a year later posting anything out of here that is fatter than 3cm is pretty hard. I think most of Europe is ok for small airmail parcels again. If you don't mind waiting I can send you a breadbee and maybe a SSD202D module.

Cheers,

Daniel

mbilloo, 2021-04-23:
> not accept the way I implemented it..

Romain,

I'd  be interested in helping you. I've been doing kernel development for
the past few years for private companies, and would like to start
contributing to OSS (I've never found the correct vehicle to do so). Let me
know if you want to split up the patches - I'd be interested in reworking
the clocks as Daniel has mentioned. If there's another way to split up the
work that makes more sense to you, please let me know - just want to
contribute somehow.

rperier, 2021-04-23:
Hi,

> Taking some of the current backlog of ~300 patches would really help.
> For example the clocks need to be totally reworked because mainline will not accept the way I implemented it..

Yes, I can help for this for sure. Yeah, get patches merged upstream takes time and efforts (but that's also super cool and you learn a lof of stuffs, like when you hack a new freash device).

> Debian is probably impossible in 64MB of RAM right? The SSD202D with dual A7 and 128MB of RAM would probably be more suitable there. They are almost the same internally so anything that helps the SSD202D helps this board and all of the other chips in this family.

For just booting linux and booting a small rootfs, 64MB of RAM is good, ihmo.For linux you need at least ~5-6MB of RAM and that's enough, I already ran a mailine kernel with buildroot on a STM32F469i-Discovery (16MB of RAM, no MMU). A mainine kernel was working like a charm. Porting an embedded debian on a new device is done via tools like debos or vmdb2 (you don't run an official debian installer).  I just need a boot medium with enough of space, so an usb boot medium would be okay or a rootfs over NFS would be okay too.

> Anyhow it seems you are in France? I have a bunch of boards made up for the people in this thread but even a year later posting anything out of here that is fatter than 3cm is pretty hard. I think [...]

Ah nice, having two modules would be cool.
Yeah I am in France (I can share my postal address via private message for more details). Of course, I can wait, no problem with that :) . If I need to contribute to postal fees or anything, feel free to ping me.

Regards,
Romain

fifteenhex, 2021-04-23:
> Yes, I can help for this for sure. Yeah, get patches merged upstream
> takes time and efforts (but that's also super cool and you learn a lof of stuffs, like when you hack a new freash device).

The base platform support is in mainline since 5.9 I think. I think I've gotten something into every release since then but it's hard work. Clocks are a real blocker now as without them all of the nice stuff like ethernet can't happen.
 
> For just booting linux and booting a small rootfs, 64MB of RAM is good, ihmo.
>For linux you need at least ~5-6MB of RAM and that's enough, 

64MB is fine for buildroot but I think Debian would really struggle. Especially since the local storage is only 16MB.

>I already ran a mailine kernel with buildroot on a STM32F469i-Discovery (16MB of RAM, no MMU).

I did that with a mc68000.. it was fun but nommu linux is awful really. Hence when I found a chip that could run proper Linux, wasn't MIPS and hand solderable I took about reverse engineering it. :)

> Yeah I am in France (I can share my postal address via private message for more details

Send me an email and we'll work it out.

rperier, 2021-04-23:
> The base platform support is in mainline since 5.9 I think. I think I've gotten something into every release since then but it's hard work. Clocks are a real blocker now as without them all of the [...]

I agree about local storage, usually for debian you need 80-100MB min for storage. For runtime, the difference at runtime will be, you use busybox by defaut with buildroot (with uclibc or musl), while it is binutils,coreutils and so in debian (well, ok, good point with a glibc). But the stuffs running at runtime being a linux kernel and few processes like systemd, getty... and a shell? But I agree that it is perhaps "sport" and hard yeah :=D 

> 
> > I already ran a mailine kernel with buildroot on a STM32F469i-Discovery (16MB of RAM, no MMU).
> 
> I did that with a mc68000.. it was fun but nommu linux is awful really. Hence when I found a chip that could run proper Linux, wasn't MIPS and hand solderable I took about reverse engineering it. :)

Hehe, nice :)

> 
> > Yeah I am in France (I can share my postal address via private message for more details
> 
> Send me an email and we'll work it out.

Done to your email @thingy .

Thanks,
Romain

fifteenhex, 2021-04-23:
> I agree about local storage, usually for debian you need 80-100MB min for storage.

Even with SPI NAND that will be pretty tight. The biggest chips that are not BGA are ~256MB I think.

>For runtime, the difference at runtime will be

Right now we get a buildroot userland with glibc (uclibc and musl both suck in different ways..) and python3 etc and a kernel + a smaller ulibc based userland and stripped down kernel for recovery situations into the 16MB of NOR flash.

```
Welcome to breadbee. Your user is 'bzzbzz'. Type 'sudo beecfg' after logging in to start configuration.
breadbee login: bzzbzz
$ uname -a
Linux breadbee 5.12.0-rc8+ #1457 SMP Fri Apr 23 00:15:32 JST 2021 armv7l GNU/Linux
$ cat /proc/cpuinfo 
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 3.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : MStar/Sigmastar Armv7 (Device Tree)
Revision        : 0000
Serial          : 0000000000000000
$ free
              total        used        free      shared  buff/cache   available
Mem:          53528        9324       39124          76        5080       40952
Swap:             0           0           0
$ python3
Python 3.9.0 (default, Nov 26 2020, 22:17:59) 
[GCC 10.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> print("yo!")
yo!
>>> 
```

rperier, 2021-04-23:
Nice ! 
Yeah at the beginning, I will probably use a compressed initramfs for developing more easily and quickly (or a small rootfs in the NOR, I don't know yet), until I get an external boot medium with enough of space ;)

fifteenhex, 2021-04-23:
I use TFTP to load a kernel for dev. For the SSD202D etc with 128MB of memory I use an initramfs inside of a FIT with the kernel but on the MSC313E with 64MB of memory you can't do that really as the decompressor for the initramfs tries to allocate 64MB of temporary space from what I remember.

fifteenhex, 2021-04-23:
Hi @mbilloo 

Your board is still packed up on my desk waiting to be sent. It's getting pretty dusty :(.

If you want to get involved in some mainline stuff you can always take a look at the commits in my queue branch here:
https://github.com/fifteenhex/linux/commits/msc313_mainlining

This is the stuff that is complete enough that it's worth sending for review. But I still have boneheaded mistakes that I missed a year and half ago and miss again when I go through the commits before sending them to mainline. If a fresh pair of eyes looked at it before I sent the first version of a series I'd probably be able to skip at least one resend for stupid stuff like bad indenting.

In that branch everything up to "dt-bindings: clk: mstar msc313 cpupll binding description" has either been queued or is "out for review". From that commit and onwards is stuff I'm checking again now to split into series and send them out.

rperier, 2021-04-23:
Yeah it was not my point, usually I use TFTP for loading the kernel too (more helpful and quicker), what I meant is for the beginning when few boot mediums are available or when few resources are available I try to use either small rootfs or small initramfs (by using a mainline buildroot or yocto, most of the time buildroot) , but, as you said, it also depends on the technical constrainst too , of course. Anyway, I will focus on kernel development first (that's the main interest, even if debian is cool, have everything supported upstream is definitively more sexy)  ;)

As you said the main focus being, the clktree driver and perhaps the pinctrl driver too nop ? I don't see it in linux-next.

Thanks,
Romain

mbilloo, 2021-04-23:
Hey @fifteenhex ,

> Your board is still packed up on my desk waiting to be sent. It's getting pretty dusty :(.

Eh, that's expected. Hopefully you'll be able to send it out towards the end of this year.

> This is the stuff that is complete enough that it's worth sending for review. 

Do you care about how it's done? I.e. create a fork of your repo, for each commit, run the standard checks, fix any issues, and submit a PR once they're all done. Or do you want me to submit a patch for each commit directly once everything checks out? I take it you're the maintainer either way, so it doesn't matter, but wanted to see if you had a preference.

fifteenhex, 2021-04-24:
> > This is the stuff that is complete enough that it's worth sending for review.
> 
> Do you care about how it's done? I.e. create a fork of your repo, for each commit, run the standard checks, fix any issues, and
submit a PR once they're all done. Or do you want me to submit a patch for each commit directly once everything checks out? I take it you're the maintainer either way, so it probably doesn't matter, but wanted to see if you had a preference.

I wonder what the best way of going about it is. I think because I'm still rebasing and squashing patches PRs will break pretty quickly. I think doing the normal kernel mail process but off of the kernel list might work better. So I send a series of patches. You get emails. You can review and reply via email and then I can do the fixes and send a new series.
Or maybe I just promise to not touch the queue branch until you've done a review and added a PR for any fixes.

rperier, 2021-04-24:
@mbilloo 
more we are, best it is ;)

@fifteenhex @mbilloo 
We can rebase our work onto the torvalds's tree, but ideally we should rebase onto either linux-next or onto the maintainer's tree found in git.kernel.org (it depends on the subsystem) before sending a series, so things will break pretty quickly, yes. 
I agree, doing the normal kernel mail process will work better and will be faster. You will also get more feedbacks.

Daniel, for future series, you send emails with us in Cc: ?

PS: I have just created an IRC channel #linux-chenxing on freenode (you can also speak with a lot of maintainer from there there is the #armlinux channel). If you prefer another discussion mechanism, feel free to let me know.

Romain

fifteenhex, 2021-04-24:
> @fifteenhex @mbilloo
> We can rebase our work onto the torvalds's tree, 

That's what I do.

>but ideally we should rebase onto either linux-next or onto the maintainer's tree

linux-next changes too much on a daily basis. I rebase onto linus' master daily and cherry pick anything from linux-next that is needed. If/when there is more than one person working in multiple feature branches for each subsystem and having all of that auto merge onto linux-next might be fun. But right now I'm one guy hacking on this code for a bit of fun.

>found in git.kernel.org (it depends on the subsystem) before sending a series, so things will break pretty quickly, yes.

At the moment the changes aren't touching any shared code. To the only chance for conflicts so far is in shared Makefiles. The ethernet and usb patches might need to be rebased onto the right tree before sending but that's a bit away.

> I agree, doing the normal kernel mail process will work better and will be faster. You will also get more feedbacks.

I'm already using the normal kernel process. Please look in the MAINTAINERS file...

This is about pre-reviewing stuff before it goes out. A lot of the patches in my tree have been around for almost 2 years now. 2 years ago I didn't plan on mainlining any of this so I didn't pay a lot of attention to doing things right so there are ident errors, checkpatch errors etc.

I don't have much time for this so I eyeball it, build it, boot it, run checkpatch on it and then send it. That results in silly issues like I wrote "__iomem void" when it should be "void __iomem" and copy and pasted that a few times and never noticed until the upstream arm soc maintainer checked. The mainline review process is partially to catch that sort of stuff but if I/we catch it before the series is ever sent it means upstream people should only ever review stuff that is free of really silly mistakes.

> Daniel, for future series, you send emails with us in Cc: ?

I don't mind doing that. But really I want to get to the point where a series goes out and it only really needs an ACK from the sub-system maintainer to get merged.

> PS: I have just created an IRC channel #linux-chenxing on freenode
> If you prefer another discussion mechanism, feel free to let me know.

I don't have a lot of time and I do not touch this stuff during work time so realtime interaction is basically zero.. so I prefer using githubs discussion feature.
https://github.com/linux-chenxing/linux-chenxing.org/discussions

I think IRC would make sense if/when there is a community of people hacking away on this. It would be really cool have a bunch of really skilled people hacking away and reverse engineering the remaining hard bits like the camera, video encoder/decoder blocks etc but I don't see that happening.

rperier, 2021-04-24:
> That's what I do.
>
>
> linux-next changes too much on a daily basis. I rebase onto linus' master daily and cherry pick anything from linux-next that is needed. If/when there is more than one person [...]

Good point.

> I'm already using the normal kernel process. Please look in the MAINTAINERS file...
>
> This is about pre-reviewing stuff before it goes out. A lot of the patches in my tree have been around for almost [...]

Well, I realize that I have read your suggestion too quickly, my bad , sorry ^^ (I missed the "*but off* the kernel mailing list" XD). So a pre-review only between us by using git send-email, right ? yes, why not

> I don't have much time for this so I eyeball it, build it, boot it, run checkpatch on it and then send it. That results in silly issues like I wrote "__iomem void" when it should be "void __iomem" [...]

I understand, this is precisely where we can help (with other improvements, if upstream asks for it)

> I don't have a lot of time and I do not touch this stuff during work time so realtime interaction is basically zero.. so I prefer using githubs discussion feature. [...]

No problem, I understand. Let's switch to github discussion then.

mbilloo, 2021-04-24:
> So I send a series of patches. You get emails. You can review and reply via email and then I can do the fixes and send a new series.

I think this is the better approach. I can start off by creating the patches from your commits, run each through the regular checks. If the patch looks good, I can just email you the patch, and you can email it to the main kernel list whenever you get a chance. If the patch needs some tweaks, I can create a patch on top of your commit, and send that to you for approval. This way, we're still closely aligned with the kernel development process, and distributing the work load over time. In my opinion, this will avoid having to do redo all of this work at the end when multiple PRs need to be reconciled (instead we would just have patches we could just summarily forward to the kernel list with some decorative changes).

I can start this with a first patch and email you and @rperier to get the ball rolling, once you give the green light.