breadbee/breadbee #26: Mainlining effort

kadamski, 2020-05-24:
I've seen you have posted first few patches to the mainline but the effort seems to be stuck for now. I guess, you have a lot of other work and I know mainlining anything requires quite some time. Is this something you would like to be helped with or this is something you would like to do yourself?

I feel like I could help out with that, once I've got a board to test the code on.. The M5 in dashcam, if I remember correctly, is not suitable for that?

fifteenhex, 2020-05-24:
TL;DR; version - I'd like help but at the moment my tree is a bit of a mess. Working on something like the aesdma that I'm not poking at right now might be easier because I won't be constantly breaking your stuff with rebasing...

Longer version;

I have a patch set that I think I can clean up and try pushing again but I haven't gotten around to finishing up the cleaning up part. Splitting out all of the device tree changes and moving them into the commits they belong with is a major pain (basically merging all the dt commits into a jumbo commit, moving it before the other commits, splitting it back out again.. )

The M5 is suitable for mainlining work I think. One of the big changes between the current code and the first version I pushed is that the current stuff should work with almost any chip in the family whereas the first attempt would have only worked with the msc313e. We should be able to push an initial patch that adds support for the msc313(e) and the m5 chips in one go.

Once I have the memory init and sd card driver stuff ported from the m5 build to the breadbee build of u-boot I'm going to get back onto mainlining. I really want to get initial support in before the first run of boards is made. Once the initial bits are in getting the clock support etc is going to be a big undertaking and it would be good to maybe give you the clock stuff to push, someone else the pinctrl stuff etc

kadamski, 2020-05-24:
> Splitting out all of the device tree changes and moving them into the commits they belong with is a major pain (basically merging all the dt commits into a jumbo commit, moving it before the other commits, splitting it back out again.. )

Seems like a work I do a lot daily so, well, I don't feel like it is a pain to me to do such thing so we could cooperate somehow on that if you like.

fifteenhex, 2020-05-29:
I'll have a think about how we can do it effectively. I probably need to clean up my work branch a bit, update the "for mainline" branch and then we can clean up all of the checkpatch etc errors and push it.

fifteenhex, 2020-06-09:
Planning to drop a patch set for the initial support into the 5.8 release cycle.

Work will be happening here: https://github.com/fifteenhex/linux/tree/msc313_mainlining

kadamski, 2020-06-09:
@fifteenhex cool. If you could CC me when you are going to push something, that would be great.

fifteenhex, 2020-06-09:
The commits I'm going to push are in place now. If it goes in it'll be the bare minimum for a shell from an initramfs on both the breadbee and the m5 camera.
If you could take a look and see if you notice anything stupid that would be really helpful.

Planning to run checkpatch etc tomorrow and send it out after that.

kadamski, 2020-06-10:
@fifteenhex there's not much of the code to look at yet. I was only able to point out some cosmetic issues so far. What I would like to do, but I'm not sure I will have time for this today, is to actually boot the camera board from this kernel.

fifteenhex, 2020-06-10:
Thanks for taking a look. The first set of patches has been pushed. Hopefully we can get those in for 5.8.

I think the next thing we should try if/when that set goes in is add the parts that are purely device tree based like reset support and the internal sram. Those would be an easy "get something in mainline" win so I'd like to hand them over to someone that wants to give it a try.

fifteenhex, 2020-07-25:
Update on this: Initial patches have been merged into an upstream branch so should eventually make it into mainline.

stappersg, 2020-07-25:
> patches have been merged into an upstream branch

What is the URL of _an upstream branch_?

fifteenhex, 2020-07-25:
https://git.kernel.org/pub/scm/linux/kernel/git/soc/soc.git/log/?h=arm/newsoc

The patches are right near the top at the moment.

fifteenhex, 2020-08-04:
Update: the initial patches entered Linus' tree today so should be in 5.9

closed 2020-11-29 by fifteenhex