breadbee/breadbee #27: MK5P USB connector sucks.

fifteenhex, 2020-06-07:
Find a connector that has through board lugs that has a generic version.

kadamski, 2020-06-26:
I've went trough different boards I have to see how they handle that and indeed most of them use microusb connector with a trough-hole soldered lugs. The bluepill boards I have are a notable exception, however, as they are soldered with single SMD pads only on both sides (like MK5P) and the only difference is they seem to have small plastic lugs going to the other side.
[image]
The only way to peal it off the board was to use leverage of the microusb cable connector. This type of connector requires much less space on the bottom layer compared to the trough hole lugs.

When I tried to pill of the connectors on my nodemcu boards which have 4 trough hole lugs, it was much much harder though, to the point I was worried I would actually damage the cable connector.

Looking at the board, however, I think it wouldn't be much of a problem to fit those through hole pads for the microusb cable but the question is how will this influence the manufacturing process. Maybe it would be best to work this out with the manufacturer and pay less attention to the wide availability of the connectors?

One more thing to note, however, is that the ETH jack is also SMD mounted only. The mounting pads are larger but the connector is too so it is much easier to use leverage on it.. if microusb is easy to peel off, wouldn't ethernet jack be susceptible to that too?

One last thing is the soldering process - maybe the reason why it wasn't that easy to peel off the connector on bluepill was because it was soldered with a high temp no lead solder while you are using low temp solder paste? Did you try to use normal solder and an iron just for the side pads of the connector to give it a little more strength?

fifteenhex, 2020-06-26:
I think it was probably the batch of connectors I had. I tried pulling the same style of connector off of some older prototypes I made and it took a lot of force to pull them off and the pads came off with the connector. The ones I used recently leave the solder intact and the underside doesn't look like the solder wetted to it.

moqmar, 2021-04-25:
What about a USB Type C version of the board? The [Molex 2171790001](https://www.molex.com/pdm_docs/sd/2171790001_sd.pdf) looks a bit stronger as the shield pins are soldered *and* go a bit into the board, and is relatively cheap. The [JAE DX07S016JA1](https://www.jae.com/en/releases/detail/id=92529) would additionally have plastic pins, but is a bit more expensive.

fifteenhex, 2021-04-27:
Going through the board is a bit of no go at the moment as it would break the routing.
I think the 16 pin USB-C connectors would be the way to go. I'm going to use one on the SSD210 board I'm working on.