breadbee/breadbee #30: Rethink "timing" capacitors

kadamski, 2020-06-16:
The board has a few RC circuits used for power-up timing. The sequence seems to be:
a) the EN1/2 of EA3036C should start first (for 3.3 and 1.8V), they need some time to stabilize
b) the EN3  of EA3036C should go at least 1ms after that (to get 1.0V VDD)
c) the reset pin should not go low (~1.3V) for at least 5ms

To achieve that, a 1uF capacitor and 1M (a), 1.1M (b) and 1.2M (c) Ohm resistors are used. I see two problems with that:
1. The typical tolerance of the cheap resistors are 5%. 1% are not much more expensive so they could be used instead. However the typical tolerance of the capacitors is 10% (or even 15%) and getting more accurate isn't that easy/cheap. With bad luck, 1M*1.1 is bigger than 1.1M*0.9 and this could cause trouble. To solve this, the resistor value differences could be bigger.
2. The datasheet of EA3036C is suggesting using 100K resistor which is an order of magnitude smaller. Since those resistors are also acting as pull-ups after reset sequence, it might make a difference (10M might be to weak pull-up?). Similarly the msc316 databrief seems to be suggesting 10k for reset pull down which is 2 orders of magnitude smaller.

While smaller pull-up/down resistors increase the power consumption, with 100k or even 10k we are still talking microamps here while the SoC itself consumes tens of milliamps so this should not be a big deal, I think. The design does already use some 100k/150k/200k resistors and 100nF capacitors so changing to those walues will also shorter the BoM list as well making assembly slightly cheaper.

fifteenhex, 2020-08-11:
From @wtarreau

So regarding the resistors choices, here's what I can propose:
  - use 300k, 680k, 1000k for en1+en2, en3, cpu_reset
    => that really cuts the en1/en2 time to 1/3, hoping it's still
       enough.

  - replace all 100k bleeding resistors with 150k

  - readjust divider values accordingly.

With this I can reduce the BOM from 11 resistor values to 7:

   R1,R19,R20 = 300k
   R2,R6,R18,R21,R22 = 150k
   R3,R5,R7 = 680k
   R4,R8,R23 = 1000k
   R9,R10 = 390
   R11,R12,R17 = 0
   R13,R14,R15,R16 = 22

This gives the following voltages:
  1.0V => 0.6*(1+680/1000) = 1.008
  1.8V => 0.6*(1+300/150) = 1.800
  3.3V => 0.6*(1+680/150) = 3.32

wtarreau, 2020-09-19:
Hey Daniel, just an update on this, I've just assembled my 3rd board using the resistor values above and it works fine, so I guess the values can be updated to ease component sourcing by end users.