breadbee/breadbee #34: [TODO] New version with SSC325 instead?

fifteenhex, 2020-07-17:
The upgrade to the MSC313E seems to be the SSC325. They are currently ~$1.80 on the market which is better than the $3.40 best price I have for the MSC313E. The SSC325 has 88 pins though so it might not fit.

kadamski, 2020-07-21:
How about its availability? Can it be bought in single quantities?

fifteenhex, 2020-07-21:
I don't think I can get singles at $1.8 each but I can get 100 at that price apparently.
Trying to find a reasonable pin out before buying 100 of them though. :)

fifteenhex, 2020-07-29:
I found a broker that had them for $1.50 each so I bought 100.

assa31, 2020-08-03:
Did you find a pinout for this chip?

fifteenhex, 2020-08-04:
> Did you find a pinout for this chip?

I have a partial pinout and it seems it's probably pin compatible with the msc313dc that I have a databrief for.

assa31, 2020-08-04:
> > Did you find a pinout for this chip?
> 
> I have a partial pinout and it seems it's probably pin compatible with the msc313dc that I have a databrief for.

Could you please share your databrief?

fifteenhex, 2020-08-04:
Databrief for the MSC313DC is here: http://linux-chenxing.org/infinity3/msc316dc_pb_v03.pdf
The pinout of the SSC325 I have part of (part of a screenshot of a schematic) matches up so it's probably mostly the same.

assa31, 2020-08-04:
> Databrief for the MSC313DC is here: http://linux-chenxing.org/infinity3/msc316dc_pb_v03.pdf
> The pinout of the SSC325 I have part of (part of a screenshot of a schematic) matches up so it's probably mostly the same.

Thank you, do you perhaps know what baud rate these chips work at? And if pin 22 and 23 ate the debugging pins?

fifteenhex, 2020-08-04:
22 and 23 are the pm uart pins which is usually used for the u-boot console etc. The normal baud rate for those is 115200.

assa31, 2020-08-19:
Does anyone know what cross-compiler the mstar uses?

fifteenhex, 2020-08-19:
If you boot one of their kernels and take a look at the kernel version string it should show what toolchain they built it with.

BTW: this isn't really the place to dig for general MStar/SigmaStar information. Especially about their SDK etc as the intention is not to use any of their code as it's total garbage.. you can look for yourself as people have uploaded them to github. Maybe for general information take a look at the linux-chenxing site I made (http://linux-chenxing.org/).

assa31, 2020-10-09:
Hi, I'm busy working on one of their camera modules. I managed to get the toolchain and build a kernel. but when I run their load ko file then I get the error unknown symbol. do you maybe know of a solution or of other camera libraries that I can use?

fifteenhex, 2020-10-09:
@assa31 

Sorry this issue isn't a support area. Please stop asking unrelated questions here.

fifteenhex, 2020-11-29:
Another reason to do this:

The SSC325 can apparently deep sleep at ~3uA.

iamdavidcz, 2021-04-16:
@fifteenhex 
I have managed to create a simple schematics of the possible candidate for the second version of Breadbee. I would like to integrate the following components on the 32 x 32 mm board:
1. SigmaStar SSC325 (rather obvious)
2. Ampak AP6256 (2.4/5 GHz dual band 802.11ac chip)
3. Winbond W25N01GVZEIG (1Gbit SPI NAND Flash)
4. EA3036C (3-channel power management IC).

Certainly, there are many mistakes and faults in this schematics considering my little experience in designing PCBs, but I would like to share what I have already achieved. Could you create a new **breadbee2** repo and give me push access to it so others could contribute too?

[image]

fifteenhex, 2021-04-16:
Hi @iamdavidcz 

I just notice I got an email from you a few days ago.. sorry for not replying yet..

Anyhow the SSC325 version of the BreadBee is/was going to be called the SuperBreadBee.
I created a repo for it a long time ago but never pushed what I had. https://github.com/breadbee/superbreadbee
For now I have made a repo for you here: https://github.com/breadbee/wip_bb2

Maybe we can make this the superbreadbee at some point.

iamdavidcz, 2021-04-16:
@fifteenhex Thanks, I will push my KiCad files to **wip_bb2** repo for now.

iamdavidcz, 2021-05-03:
Hi @fifteenhex,
I have sent you a repo with KiCad project for a new Breadbee.
Please take a look when you get a minute.

fifteenhex, 2021-05-03:
Hi @iamdavidcz 

I'm going to take a look tomorrow. If I don't spot anything obvious I'll make an order on oshpark.

iamdavidcz, 2021-05-09:
Hi @fifteenhex,
Did you get a chance to look into my design files?

fifteenhex, 2021-05-10:
Hi @iamdavidcz 

Sorry I got pretty sick over the weekend and didn't get around to looking at it. I'll try to take a look tomorrow.

iamdavidcz, 2021-05-10:
@fifteenhex
There's no hurry, I hope you feel better now :)

dasfdlinux, 2021-08-11:
@iamdavidcz 
I don't see your design files on the wip_bb2 or superbreadbee repos anywhere. Do you have them on one of these repos that I can checkout and take a look at? I'm curious about the project, and was thinking about looking into the future version of the hardware.

iamdavidcz, 2021-08-19:
@dasfdlinux
They are located [here](https://git.iamdavid.cz/breadbee2/).

@fifteenhex
Feel free to push them to wip_bb2 repo :)