breadbee/breadbee #38: 2 microphones through ALSA (assuming working on-board audio coder)

dumblob, 2021-11-04:
Just discovered this very nice SoM which is one of the closest to what I'm looking for.

I'm interested in Linux (incl. ALSA) capable as-small-as-possible board with an integrated (on-board or in SoC/SoM) audio codec which supports two microphones.

Does this board offer 2 mics? Does the ALSA support already exists for the codec?

If not, do you know of some other very small Linux-capable (64MByte RAM is enough) SoM or PCB with integrated codec for 2 microphones?

----

As a nice to have would be support for battery power supply and an LTE (I need full speed and not LTE-M) shield. But that's lower priority than two mics working with ALSA out of the box.

HeinzAlberto, 2021-11-04:
Hi there!

I'm sorry but I don't think that you'd be able to connect 2 mics. The SoC has one ADC for a mic (as well as an DAC for mono line out).
Anyway, there are other interfaces. Guess you could use SPI (some MEMS support it) or go over the USB 2.0 host interface and use a external sound chip...

- Hope that helps you!

Btw.: SD-Boot ist supported by the SoC

HeinzAlberto, 2021-11-04:
(If you're willing to code a bit more you can also check out the recent ESP32-variants. If the are able to do multi-mic VoIP-Calls woth active noise-cancelling, it might also be able to solve your problem ;-) )

dumblob, 2021-11-04:
Thanks for clarification about the number of ADCs. USB is a bit off for my use case due to high power consumption. But it'd be the easiest solution, that's for sure.

> (If you're willing to code a bit more you can also check out the recent ESP32-variants. If the are able to do multi-mic VoIP-Calls woth active noise-cancelling, it might also be able to solve your problem ;-) )

I need to use opus to achieve the highest quality while maintaining acceptably low bitrate. So if modern ESP32 is capable of encoding to Opus in near-realtime (1 second delay would be acceptable), then I'd consider writing more code and go that route instead of fully Linux-capable SBC/SoM. But I slightly doubt it's fast enough (considering also the limited RAM options I saw with ESP32 boards).

I'll continue with the hunt then. Thanks anyway!

(actually I'm getting a bit desperate, so NDA will probably also be an option - IDK yet)

dumblob, 2021-11-04:
Oh, btw. I don't have any experience with MEMS mics especially when it comes to their quality. Do you think they offer good quality say even for some occasional all-around music recording (not just speech)?

closed 2021-11-04 by dumblob

HeinzAlberto, 2021-11-05:
I don't knoe how they perform with really low frequencies but in general there are many MEMS out there with good quality. Okay, every Smartphone has one nowadays but speaking from e.g. my 2019 semi-professional camcorder, it has a MEMS too (and it sounds great, even on concerts).
Of course you can't really compare it to these ECMs RØDE uses but compared to typical ECMs I would consider MEMs at least "good as". (And they have a higher SNR.)

fifteenhex, 2021-11-05:
I think the biggest roadblock here would be that the audio part of these chips isn't worked out yet.
The SSD201/SSD202D apparently support 4 dmics but that's not much use without an ALSA driver.

dumblob, 2021-11-07:
> I don't know how they perform with really low frequencies but in general there are many MEMS out there with good quality. Okay, every Smartphone has one nowadays but speaking from e.g. my 2019 semi-professional camcorder, it has a MEMS too (and it sounds great, even on concerts). Of course you can't really compare it to these ECMs RØDE uses but compared to typical ECMs I would consider MEMs at least "good as". (And they have a higher SNR.)

I need to buy some of the best ones and try - internet seems generally positive about the development of MEMS mics (mainly driven by smartphones).

> I think the biggest roadblock here would be that the audio part of these chips isn't worked out yet. The SSD201/SSD202D apparently support 4 dmics but that's not much use without an ALSA driver.

Yep, that'd be the biggest roadblock. Basically I wanted to avoid developing yet another shield with an audio codec. I've done that once (PCB, DTB, and quite extensive driver "glue" as there was some generic driver just for the codec) but for a second version of the system I just thought why not to stick with an existing platform. Apparently it's not that easy to find such platform :wink:.

dumblob, 2021-11-08:
> It's moving away from the breadbee, but about an RPi Zero? It should have enough power by far to drive serveal mics, it has good documented hard- and software. (If thats not to expansive)

Yep, that'd be totally fine if I wasn't this much constrained by physical dimensions. Breadbee is less than half the size.

HeinzAlberto, 2021-11-08:
Than I could offer VoCore or, a bit bigger, NanoPi Neo

dumblob, 2021-11-08:
>Than I could offer VoCore or, a bit bigger, NanoPi Neo

I'm in contact with VoCore guys - they said it's possible but not out-of-box prepared for 2 mics (non-MEMS). But if the work would be upstreamable, I'd consider it - I just didn't get any clear answer whether they would accept such patches and if it's even possible to make the patches "backwards" compatible (i.e. without disturbing anything in the existing ecosystem).

NanoPi Neo should be acceptable if used with strict low-power settings (almost everything turned off and CPU quite heavily underclocked) but I'd need to de-solder the USB & ETH connectors (in case of breadbee there seemed to be a choice to just "leave the ETH connector out" during pick & place).

fifteenhex, 2021-11-09:
@dumblob 

There are some very small and cheap ($10) SSD202D modules out there. That would give you 2 x cortex A7 and 128MB of RAM and apparently support for multiple dmics. If you know audio stuff maybe you could help me writing a driver for the audio hardware?

dumblob, 2021-11-09:
> There are some very small and cheap ($10) SSD202D modules out there. That would give you 2 x cortex A7 and 128MB of RAM and apparently support for multiple dmics.

Yep, that's more or less why I considered breadbee in the first place :wink:.

>If you know audio stuff maybe you could help me writing a driver for the audio hardware?

I wouldn't say I know audio stuff - it was anything but smooth work for me to write all the thousands of lines of highly error-prone code with basically zero chance to test it efficiently (this is the main pain point except for the enormous time sunk in the development itself). With audio it's very common to find mistakes first after many hours of error-less operation unlike with most other peripherals which either work first time and then forever (with minor exceptions due to different conditions of someone elses setup). And there are usually tens of such mistakes. It's probably the most tedious and dissatisfying work I've ever done (and I've done even crazy stuff like e.g. *marcan* does with the new Apples though in my case in much smaller scale). Which might explain why audio is either not being supported on SBCs/SoCs only partially or often not at all (btw. Linux internal audio interfaces changed quite significantly during the last 5 years - so even plain maintenance of a driver is PITA).

And that's why I rather devote the time to browsing and looking for an already working solution. I was even looking up codec chip which could be connected through **eUSB** and had 2 mics (not dmics) because that'd solve the issue for me - it's almost as low power as I2S+I2C connected codecs but with 0 development time because it's standard USB audio. Unfortunately I couldn't find any.

All in all thank you for asking me to help out, I feel honoured. I would very much like to say *yes* though I'm not yet in the position to devote my time to such a rabbit hole.

-------
Maybe it's time to create & sell a miniature "bridge" (ideally an ASIC or some DSP/uC but maybe even FPGA...) converting some ultra low power I2S+I2C codec (featuring at least 2 mics, several dmics etc.) to standard USB audio and offering eUSB instead of standard USB, so that integrators can just place it on PCB to have **0** effort full-featured audio with more or less the same low power consumption as a "native" codec (and those SoCs without eUSB would use a cheap miniature eUSB<->USB bridge which is generally well available).