breadbee/breadbee #43: Freshly built doesn't boot

stonehencs, 2024-09-29:
Hello,

I'm so desperate that I had no other idea where to ask for help, so I'll try here.

I built a breadbee by myself and everything is in place/all soldered perfectly and all volts are there. Even windows recognizes the CH340E but no data is coming in on UART. I tried with different SPI NOR chips finally I stayed with W25Q256FV because of the large memory capacity of the chip. I even managed to run buildroot for the 5th time to generate the image which I was able to write to the SPI chip.

But after that I don't have much idea what could be wrong.
Anyone have any ideas?

Thanks in advance!

fifteenhex, 2024-09-29:
Congrats on building one! If you set the baud rate for your serial to 38400 (I think) do you see anything? The boot rom in the chip if it is working will print out some debug stuff.

I haven't touched this stuff in ages but I'll help you to get it going.

stonehencs, 2024-09-29:
First of all, thank you very much! 
I tried but got nothing. At the moment I suspect that the crystal is not working because I tried 2 MSC313Es but they all had this problem.
I'll attach two pictures of how I did it, but first I had no way to write the SPi chip itself on the crib board so I improvised with a gold connector and some cable.

[image]
[image]

fifteenhex, 2024-09-30:
If you stick a multimeter in voltage mode on the crystal do you see anything?

stonehencs, 2024-09-30:
I can't see anything, so I suspect that's where the problem might be. But this is my last guess.

fifteenhex, 2024-09-30:
> I can't see anything, so I suspect that's where the problem might be. But this is my last guess.

Ok, you should check you are getting the correct voltages on the different rails.

stonehencs, 2024-09-30:
I have checked and there is the right voltage everywhere.

I ordered a new crystal, as soon as it arrives I will replace it and report back. Should be here in 2-3 working days.

stonehencs, 2024-10-07:
Sorry for the delay, I got the crystal, I managed to put it on and now I have about 1.5V there but it still won't start and I get nothing on UART. I checked all the voltages again and everything is there, the machine recognises the CH340 as before but spits out nothing.

stonehencs, 2024-10-09:
@fifteenhex I was wondering if there's any way I could measure if the chip is working?

fifteenhex, 2024-10-10:
> @fifteenhex I was wondering if there's any way I could measure if the chip is working?

Give me a day or so to get back to you. Work is crazy at the moment.

stonehencs, 2024-10-10:
Feel free and thank you for spending your free time on my problem.

fifteenhex, 2024-10-19:
@stonehencs 

I think if you are seeing ~1.5v the oscillator is running. Try connecting minicom at 38400 baud and while it is connected to usb manually reset the chip by connecting a wire to the point between c37 and r23 (the reset circuit) and poking it on a 3.3v connection. You should see characters come out and that'll tell you what is wrong.

Also if you want to drop me a private email I can probably send you a working board so you can compare against yours.

stonehencs, 2024-10-20:
Unfortunately I have a windows computer so I tried what you described with a program called realterm. But it did not detect anything even though I tried many times.
I was thinking that the chip itself might be bad and then I made a breakout board (which hasn't arrived yet) and put together the basics based on your schematic, i.e. the SPI chip, the power sources and the decoupling capacitors. Maybe it will produce something.

I will gladly accept your offer if you really have a working piece. I've even thought that if you have the time and the capacity, I'll send you my board and see if you have more advanced tools to debug it.

My private email address is |hencs.bence@gmail.com| so if you saw the post I'll delete it from here.

[image]
[image]

pak-man, 2024-10-20:
> My private email address is ***@***.***| so if you saw the

Nowadays doing with electronics is paradise: basic tools are dirt cheap,
boards also, it´s easy to program, there are zillions of resources online.

I can only recommend buying few basic tools when working with digital
electronics: a dirt cheap logic analyzer can be had for less than 10$, even
as cheap a FIVE bucks. Ebay or aliexpress with "logic analyzer 24MHz 8
channels" and you´ll find a little jewel that does not break the bank. Use
some opensource software with it like pusleview (win/mac/linux) and go on
youtube for some hands-on learning.

You´ll learn a lot, you´ll spot problems, it´s worth every cent and minute
of your time. Sure it´s limited to 24MHz, but just define a slower clock
when you initialize your SPI bus, and it´s ok!

(My first logic analyzer weighted easily 30kg, took half of my desk, and
costed me a whole two months of summer job money - second hand)

stonehencs, 2024-10-22:
@pak-man 
You're absolutely right, I was thinking mainly of an oscilloscope, but you're right that I can use a cheap logic analyzer to see if there's any movement on the SPI channel and UART.
Thanks for the idea!

fifteenhex, 2024-10-22:
@stonehencs one of the cheap analyzers that work with sigrok would be enough to check if its doing anything. But the oscillator running is a sign that the chip is doing something.
I send you a mail in a few days about sending a finished. On a trip at the moment.

stonehencs, 2024-10-30:
Does it have to be sigrok? Because the shop where I shop has one that is Saleae compatible.
link: https://www.hestore.hu/prod_10035503.html

pak-man, 2024-11-01:
It´s the same hardware, and feel free to use it with whatever software
works for your OS

stonehencs, 2024-11-03:
Okay, thank you. I've ordered it, and I'll let you know when it arrives and I've managed to make something with it.

stonehencs, 2024-11-09:
The loogic analyzer arrived and I used it right away after a few test runs on the arduino. I got quite interesting stuff I think although not so much unfortunately I don't understand it. I will upload the pictures here and leave a link where you can download the data as csv or open the session file with Saleae Logic.

I connected a channel to each data leg of the Spi chip, the crystal and the uart tx/rx output leg. I noticed that the crystal gives an output that is when it is 24mhz and when it is 12 mhz( you will see it better in the picture). On the uart concretely there is a tiny movement. This is what I was able to get some ASCII out of it into Realterm: CE0EFE3E0EC0CE003000 and it repeats this after every reset.
Link: https://drive.google.com/drive/folders/1Rbi9VHoY2qpa1jU66Z_scIadBhbOR_HJ?usp=drive_link

[image]
[image]
[image]
[image]
[image]
[image]

stonehencs, 2024-11-11:
I managed to do the break board version with the basics (i.e. the crystal, reset, and basic current rails) but I get the same ones as I posted in the previous post.

fifteenhex, 2024-11-14:
I'll try to send a dump of a working board you can write to your flash this weekend.
It does look like it's accessing the SPI flash though and that sort of shows it's working.

stonehencs, 2024-11-16:
I managed to get it to boot!!!! I regenerated the whole thing from scratch on a separate virtual machine and one of the LEDs lit up. However, my uart access was not easy because I had no normal baudrate so I had to measure it and ended up with 50253 baudrate. Finally I got this:

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2024.11.16 08:48:18 =~=~=~=~=~=~=~=~=~=~=~=
ćPp2főSF: Detected w25q256 with page size 256 Bytes, erase size 4 KiB, total 32 MiB
device 0 offset 0x80000, size 0x300000
SF: 3145728 bytes @ 0x80000 Read: OK
## Loading kernel from FIT Image at 22000000 ...
   Using 'breadbee' configuration
   Trying 'kernel@0' kernel subimage
     Description:  unavailable
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x220000a8
     Data Size:    2239360 Bytes = 2.1 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x22800000
     Entry Point:  0x22800000
     Hash algo:    crc32
     Hash value:   a73bf5ef
     Hash algo:    sha1
     Hash value:   345556c8ee516976d11b7ce3c4193588d06eb308
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Loading fdt from FIT Image at 22000000 ...
   Using 'breadbee' configuration
   Trying 'fdt@0' fdt subimage
     Description:  unavailable
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x22222d28
     Data Size:    34657 Bytes = 33.8 KiB
     Architecture: ARM
     Load Address: 0x23000000
     Hash algo:    crc32
     Hash value:   95b04d11
     Hash algo:    sha1
     Hash value:   3d326e97975ad95a50f1ad43da910bb049e309aa
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Loading fdt from 0x22222d28 to 0x23000000
   Booting using the fdt blob at 0x23000000
   Loading Kernel Image ... OK
   Loading Device Tree to 23f85000, end 23f90760 ... OK

Starting kernel ...

[    0.024239] gpiochip_add_data_with_key: GPIOs 480..511 (gpio-msc313e) failed to register, -517
[    0.137064] debugfs: Directory '1f200400.bdma-msc313-bdma' with parent 'regmap' already present!
[    0.181054] dummy-irq: no IRQ given.  Use irq=N
[    0.236241] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
[    0.244535] CPU: 0 PID: 1 Comm: swapper Not tainted 5.4.0 #1
[    0.250147] Hardware name: MStar Infinity (Device Tree)
[    0.255355] [<c010af04>] (unwind_backtrace) from [<c0109910>] (show_stack+0x10/0x14)
[    0.263043] [<c0109910>] (show_stack) from [<c011e0c8>] (panic+0xd0/0x2c4)
[    0.269871] [<c011e0c8>] (panic) from [<c0801414>] (mount_block_root+0x238/0x2d4)
[    0.277297] [<c0801414>] (mount_block_root) from [<c0801674>] (prepare_namespace+0x144/0x184)
[    0.285758] [<c0801674>] (prepare_namespace) from [<c0520ee0>] (kernel_init+0x8/0x108)
[    0.293617] [<c0520ee0>] (kernel_init) from [<c01010d8>] (ret_from_fork+0x14/0x3c)
[    0.301124] Exception stack(0xc3833fb0 to 0xc3833ff8)
[    0.306136] 3fa0:                                     00000000 00000000 00000000 00000000
[    0.314248] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    0.322359] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    0.328925] Rebooting in 10 seconds..

Anyone have any idea why gpio won't register?

fifteenhex, 2024-11-17:
That's great news. I think what you should do is stop it auto booting at u-boot and remove the quiet command line option (should be in bootargs). That way you'll see all of the kernel output. Now you have it running Linux fixing software issues should be pretty easy.

stonehencs, 2024-11-17:
@fifteenhex Honestly, I'm a beginner in this deep level of config, maybe if you could help me with setting these up I would really appreciate it. I tried to search the internet how to do it but I couldn't find anything or only what works after booting.

stonehencs, 2024-11-20:
@fifteenhex I managed to do pretty much what you recommended after about 16 hours of searching the internet and files. I rebuilt the base image, now it gives a kernel panic for a completely different error.
Log: 
[putty.log](https://github.com/user-attachments/files/17832310/putty.log)

fifteenhex, 2024-11-21:
> @fifteenhex I managed to do pretty much what you recommended after about 16 hours of searching the internet and files. I rebuilt the base image, now it gives a kernel panic for a completely different error. Log: [putty.log](https://github.com/user-attachments/files/17832310/putty.log)

Does it give you the same panic every time? If it's changing slightly each time its possible the DDR supply isn't quite right. But I think you are pretty close. If you jam a key when it reboots to u-boot it should stop at the u-boot prompt. You should be able to use ethernet there? If so we can use that to upload a different kernel etc and debug the software issues.

stonehencs, 2024-11-26:
@fifteenhex  Can you send me an SPI dump that's already booted to see if I'm sure I have a software bug?

stonehencs, 2024-11-28:
@fifteenhex I asked around on Reddit and it was brought to my attention that a file in the boot log is writing an error and that might be the problem.

`[    0.263971] kernel BUG at crypto/algapi.c:461!`

Here is the section from the algapi.c:

```
void crypto_unregister_alg(struct crypto_alg *alg)
{
        int ret;
        LIST_HEAD(list);

        down_write(&crypto_alg_sem);
        ret = crypto_remove_alg(alg, &list);
        up_write(&crypto_alg_sem);

        if (WARN(ret, "Algorithm %s is not registered", alg->cra_driver_name))
                return;

        BUG_ON(refcount_read(&alg->cra_refcnt) != 1);  <-- This is what the boot log gives an error for
        if (alg->cra_destroy)
                alg->cra_destroy(alg);

        crypto_remove_final(&list);
}
EXPORT_SYMBOL_GPL(crypto_unregister_alg);
```

Can you help me with what might be causing the error?

fifteenhex, 2024-11-30:
> @fifteenhex I asked around on Reddit and it was brought to my attention that a file in the boot log is writing an error and that might be the problem.
>
> `[ 0.263971] kernel BUG at [...]

I'm going to setup a board tomorrow and try to reproduce what is going/dump a working kernel etc for you.

fifteenhex, 2024-12-01:
> @fifteenhex Can you send me an SPI dump that's already booted to see if I'm sure I have a software bug?

Here is a dump of the flash from a board that is on my desk and working.

[bb_working_20241201.bin.zip](https://github.com/user-attachments/files/17968437/bb_working_20241201.bin.zip)

stonehencs, 2024-12-01:
Thank you very much, I will try it tomorrow and let you know what happened.

Do you have any idea why nothing changes when I change the uboot config (I just want ctrl+z to enter the uboot shell)

fifteenhex, 2024-12-01:
> Thank you very much, I will try it tomorrow and let you know what happened.
> 
> Do you have any idea why nothing changes when I change the uboot config (I just want ctrl+z to enter the uboot shell)

Try typing `bzzbzz` instead. I have a feeling the buildroot setup generates a u-boot with a boot string instead of just any key as sometimes the usb->serial chip was outputting a byte at power up.

stonehencs, 2024-12-02:
I tried, but it didn't boot either. But now it gives a new error, that it is unable to mount the root fs.

log: 
[putty.log](https://github.com/user-attachments/files/17983088/putty.log)

stonehencs, 2024-12-07:
@fifteenhex I had an idea that maybe it was the SPI chip I was using that was defective so I ordered the same one as the one in the components layout and once I ordered it I ordered another kind of crystal.
I put it together last night and guess what with the .bin file you sent me: it booted up!!!!

I tried with the image I generated but got the same error: `kernel BUG at crypto/algapi.c:461!`.

Were you able to find out why it would throw this error?

stonehencs, 2024-12-17:
@fifteenhex I managed to generate a working firmware after about 3 different ubuntu versions. I tried to add an Apache in the config but got the error that it can't find GLIBC_2.33. I checked and it uses an older version in buildroot. I got the same when I tried to add a simple custom package that would just print "Hello World". And after I took them out to see if it would run without it then it got stuck with the GLIBC error. 

How can I update or remove the error?
Boot log for the working version: 
[putty (2).log](https://github.com/user-attachments/files/18168342/putty.2.log)

Would you mind checking that I have configured the package correctly?
Link to the repo: https://github.com/stonehencs/costum_package

fifteenhex, 2024-12-24:
> @fifteenhex I tried to connect to it with ssh but it gives this error:
> 
> ```
> $ sudo dropbear -F -E -R
> [207] Dec 23 10:48:37 Not backgrounding
> [208] Dec 23 10:48:49 Child connection from 192.168.0.118:43450
> [208] Dec 23 10:48:49 Failed writing file /etc/dropbear/dropbear_ecdsa_host_key.tmp208: Invalid argument
> [208] Dec 23 10:48:49 Exit before auth: Couldn't read or generate hostkey /etc/dropbear/dropbear_ecdsa_host_key
> ```
> 
> I've looked around on breadbee but I can't find a key anywhere, and I can't generate it because it's sqfs. Is it enough to generate it with ssh-keygen and copy it there as an overlay?
> 
> I tried to play with the gpio's a bit but somehow there are pins that don't react and float between the two states. I tried to set it via beecfg but after saving and rebooting it specifically bricked. These are the pins:
> 
> ```
> gpiochip0 - 2 lines:
>         line   0:   "pm_gpio4"       unused   input  active-high
>         line   1:  "pm_sd_sdz"       unused   input  active-high
> gpiochip1 - 4 lines:
>         line   0:  "sar_gpio0"       unused   input  active-high
>         line   1:  "sar_gpio1"       unused   input  active-high
>         line   2:  "sar_gpio2"       unused   input  active-high
>         line   3:  "sar_gpio3"       unused   input  active-high
> ```
> 
> What can I do to control them?

The buildroot setup should give you a private key on your build machine and that setup on the board. I haven't built it in a while.
For gpio you should see another chip that is the main gpio controller too...
I'm going to build stuff again and try it out and see what state I let it in. I would have done it sooner but I've been busy and my PC has been dying.

stonehencs, 2024-12-24:
Sorry to hear what has happened to your computer? 
I've noticed that every time I clone from github, make finds different commands when I tab, and even when I manage to build apach2, it doesn't install on the system because I can't find any files related to it.

stonehencs, 2025-01-07:
@fifteenhexI'm trying to write a C application where I control the gpio in the memory address but I can't find the right value anywhere. I tried to find out from here: 

```
$ cat /proc/iomem | grep -i gpio
1f001e00-1f001fff : 1f001e00.gpio_pm
1f207800-1f2079ff : 1f207800.gpio
```
but it didn't work for my program. 
My code: (sorry for the language)
(https://github.com/stonehencs/costum_package/blob/main/br2apps/package/bence/raw/gpio_teszt.c)

Can you help me what it is or where I can find it?

stonehencs, 2025-01-17:
Sorry to bother you, but did you manage to build the image file or rebuild the bug? Any idea how to find the memory address?

Thank you in advance

stonehencs, 2025-01-23:
@fifteenhex I tried, especially now that I know that's the official gpio address. But I can't get it to work, it always throws an error on nmap call or memory errors. Maybe you could send me some example code or a web page with this documented?
Link my code: https://github.com/stonehencs/costum_package/blob/main/br2apps/package/bence/raw/gpio_teszt.c

 My other question is that I tried to add a read/write partition and found that it already automatically creates one and mounts it in rw directory. However, if I create one and then try to copy it to the rw partition in genimage.cfg, it will find the files I put in it, but it throws this error:

```
jffs2: error: (72) jffs2_build_inode_fragtree: Add node to tree failed -22
jffs2: error: (72) jffs2_do_read_inode_internal: Failed to build final fragtree for inode #2: error -22
```

fifteenhex, 2025-01-24:
I'm not sure about the rw partition issue but I think the GPIO one is one of the options in the kernel config like `CONFIG_STRICT_DEVMEM`.

stonehencs, 2025-01-27:
So I just enable it and I should have proper access to the memory?

However, do you remember where you created the rw partition? Or just by fstab and from the dts partition? I mean, is it created by the system at boot time and not during a build?

fifteenhex, 2025-02-01:
The r/w partition should be in the dts under the nor flash node and I think there is a script that is run on boot in /etc/init.d/ that tries to mount it (and maybe run the jffs2 format thing if needed?).

stonehencs, 2025-02-06:
I couldn't find anything there except a script to create the /rw/dropbear directory. I found that it is also set to automount in fstab.

fifteenhex, 2026-05-11:
The whole beecfg thing is a bit janky/unfinished but basically its creating a list of overlays for u-boot to apply and saving that into an environment variable. So if even u-boot can't write it that would explain a lot.. I'm thinking that maybe your flash has write protect on some of the blocks? I think there is a command in u-boot to remove that. `sf protect unlock <offset> <len>` according to a quick google.